espressif/esp-nimble #3: ble_hs_lock during bonding

BrianPugh, 2019-12-17:
During bonding, the writing portion ( `ble_store_write()` ) is surrounded by a `ble_hs_lock()` which explicitly doesn't allow nested locks.

https://github.com/espressif/esp-nimble/blob/nimble-1.2.0-idf/nimble/host/src/ble_store.c#L54

However, when it's called during bonding here:

https://github.com/espressif/esp-nimble/blob/nimble-1.2.0-idf/nimble/host/src/ble_gatts.c#L1706

It's also surrounded by `ble_hs_lock()` , so this call will always fail. Should `ble_store_write()` invoke `ble_hs_lock_nested()` instead? Or is something else wrong?

prasad-alatkar, 2019-12-17:
Hi @BrianPugh , if you look at `ble_hs_lock()` carefully it calls `ble_hs_lock_nested()` internally if BLE host debug (`BLE_HS_DEBUG`) flag is not 1, so there should not be any problem, I have verified from my end.

However I do agree that `ble_hs_lock` API's description _' Locks the BLE host mutex.  Nested locks not allowed'_ is little misleading.

prasad-alatkar, 2019-12-17:
@BrianPugh, thank you for reporting the issue, there is upstream PR which addresses your concerns,link to PR: [PR!567](https://github.com/apache/mynewt-nimble/pull/567) . I will get this merged as soon as the upstream PR is merged.

BrianPugh, 2019-12-18:
upstream was merged!

dhrishi, 2020-02-10:
Closing as this is now merged in esp-nimble as well.

closed 2020-02-10 by dhrishi