espressif/esp-nimble #10: NimBLE client cannot reconnect to bonded peer when both devices use RPA.

h2zero, 2020-04-27:
Easy to reproduce, if a NimBLE client device enables RPA and bonds with a peripheral using RPA they will connect and pair/bond. However if one of them disconnects, the client will try to reconnect using the peripheral ID address, which the peripheral does not respond to.

Commenting out [this section](https://github.com/espressif/esp-nimble/blob/0a1604a5792d64e2fc5860bfeeabf370a9680656/nimble/host/src/ble_hs_hci_evt.c#L441) resolves the issue but does not allow the client/app to obtain the peer ID address for identification.

prasad-alatkar, 2020-04-29:
Hi @h2zero Thank you for the issue. I could observe the issue and yes it gets resolved with removing code block wherein we replace the `peer_addr` in ADV_REPORT with its stored Identity address. This should be handled in `BLE_HCI_LE_SUBEV_ENH_CONN_COMPLETE`. Can you please make the update existing PR(https://github.com/espressif/esp-nimble/pull/7) with removing the above mentioned code block ?

> but does not allow the client/app to obtain the peer ID address for identification.

I could not face this issue. I could see ID addresses being updated in `LE connection complete event`. Please let me know if I am missing anything.

h2zero, 2020-04-29:
What I meant by that was in the scan discovery event the client would not be able to identify the peer by ID address. 

For instance if connection is lost and we were trying to reconnect. While scanning for that device it may have changed its advertising address and if we don’t resolve it to an ID we wouldn’t know if it’s the correct device. 

I assume that’s the reason the code exists?

closed 2020-05-22 by espressif-bot