espressif/esp-nimble #14: Connect only to bonded devices

sicanins, 2020-08-01:
Hi,
sorry for posting here, but I am trying this alread for days now.
Using nimBLE as peripheral  and it basically works fine. Appreciate that it can resolve the random addresses now!

But one important detail I just cant figure out: 
my device can be put into a pairing state where it should accept all requests and bond to new devices. 

It does that currently. 

But how do I avoid pairing/bonding to now devices while I am not in that state? It would be good to allow connections only to previously bonded devices.

If that is not possible, i might be able to solve this by OOB. But also I can not find any info if that works at all or even an example on how it would work.

Any help would be greatly appreciated!

I am on the 1.2.2 currently. Pulled the esp idf master branch and replaced the nimble folder with this repro.

Thanks!

prasad-alatkar, 2020-08-03:
Hi @sicanins 
>But how do I avoid pairing/bonding to now devices while I am not in that state? It would be good to allow connections only to previously bonded devices.

Does the device/ESP32 has input/output capabilities i.e. keyboard or Yes/No ? If that is the case then you can reject new pairing requests. If the device/ESP32 has no I/O capabilities  i.e. it only supports JUSTWORKS then it is kind of difficult to avoid new pairing requests.  

For OOB, I will get back to you in sometime regarding this. However it would be helpful if you can elaborate on how does the OOB can help you here?

sicanins, 2020-08-03:
Hi @prasad-alatkar ,

thanks for getting back to me.
The device is currently using the just works. It does have one button though.

The OOB is another idea, because I do have another communication protocol on another frequency.
At least it would be enough to pair only one device manually, and the others automatically by OOB.

Does the stack support to switch the security from just works pairing to OOB pairing only?

Even if not, I might go with a "static" key for that initial bonding during that special mode the device is in, use it to exchange more secure keys then. Not perfect but a last resort. 
My understanding is, that just works in LE Secure mode is better that that proposed idea, correct?

Looking forward to more information on the OOB.

Thank you!

prasad-alatkar, 2020-09-22:
Hi @sicanins sorry for delayed response, as I understand you want to accept/reject pairing even when in JUSTWORKS depending upon the "state". I was working on a PR to accept/reject pairing request, that may help you here, recently I have not been able to allocate time to complete that PR , however I plan to complete the PR in week's time. Will update here once done.

sicanins, 2020-09-22:
Hi @prasad-alatkar  thanks for getting back!
Actually I found that PR already and merged it, works like a charm.
Will you be doing still major changes?

Thank you!
Simon

prasad-alatkar, 2020-09-23:
Hi @sicanins Glad that my PR could help you. However if you follow the discussions on that PR, you will see that NimBLE folks have requested few changes, which are valid indeed. I will be doing those changes and accordingly will update you. It may take a week or two though !!

sicanins, 2020-09-23:
Hi @prasad-alatkar,

sounds wonderful. I followed the discussion but wasnt really sure in the end what changes they requested. 
Would be happy to change to that new version and confirm that its working!

Thanks,
Simon

rahult-github, 2023-01-11:
Hi @sicanins , its been quite some time. And the code being referred to has actually gone through some modifications. Do you still see any issue ? or this ticket can be closed ?

sicanins, 2023-01-11:
Hi @rahult-github,
we solved it by patching the NimBLE code back then. 
Means this issue is resolved for us, but maybe not for everyone else running into the same problem.

Thanks,
Simon

rahult-github, 2023-01-12:
Hi @sicanins  , thanks for the update. Closing the ticket

closed 2023-01-12 by rahult-github