espressif/esp-nimble #17: Compilation Bug with C++ in os.h

Francois-Belanger, 2020-10-03:
/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/sstream.tcc: In member function 'virtual std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::int_type std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::overflow(std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::int_type)':
/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:34:19: error: expected unqualified-id before '(' token
 #define max(a, b) ((a)>(b)?(a):(b))
                   ^
/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:30:19: error: expected unqualified-id before '(' token
 #define min(a, b) ((a)<(b)?(a):(b))

Found this:
https://stackoverflow.com/questions/6379716/c-compilation-what-did-i-do-wrong

prasad-alatkar, 2020-10-05:
Hi @Francois-Belanger recently I had also encountered this problem, `#define min` & `#define max` before including NimBLE header files (in CPP source/header files) and then undefine them might help you resolve the issue. For example,
```
#define min
#define max
#include "host/ble_hs.h"
#undef min
#undef max
```

Francois-Belanger, 2020-10-05:
Ok, I will do that, but will this be fixed ?

Also, can you please look again at the log for issue: Nimble BLE Cached access crash (IDFGH-4052) #5923

It's nasty :-). I'm wondering if it's a setting.. I don't know where to look.

Thanks

prasad-alatkar, 2020-10-20:
Hi @Francois-Belanger I had attempted to fix this issue upstream here: [min-max-macro](https://github.com/apache/mynewt-nimble/pull/866), however it won't be accepted upstream for valid reasons. So I may need to spend some time to understand and fix this issue in acceptable manner, and it may take some time :sweat_smile: . Did the above mentioned hackish suggestions help you resolve the issue ?

P.S.: Just FYI, I could find a research paper from 1995 on this "min max macro" thing, here is the link : [min max and more](https://www.aristeia.com/Papers/C++ReportColumns/jan95.pdf)

Francois-Belanger, 2020-10-20:
Hi @prasad-alatkar , yes changing includes order did fix the issue.  I hope you can fix this :-). I must say that I prefer C++ now for ESP32, much simpler to create complex app !  Thanks for the time.

okaestne, 2020-11-15:
Also had loong gcc error messages in the Pinetime (c++) project because of these. I would suggest renaming the macros to be in CAPS, so they don't collide with c++'s <algorithm>

rahult-github, 2023-01-11:
Hi @okaestne  ,  the min / max defination conflicts with other open source library also, since it is a very common name. So recently a change was done to handle the same. Can you please confirm if you observe issue even with latest IDF code ?

rahult-github, 2023-03-01:
Closing issue. Please feel to reopen, if issue still seen.

closed 2023-03-01 by rahult-github

letusgitcommit, 2024-10-30:
Still seeing this bug with min max after #include "host/ble_hs.h"