espressif/esp-nimble #18: blehr works on ESP32-WROOM but fails on ESP32-CAM

ffrige, 2020-10-17:
Sorry for cross-posting this issue, but I did not receive any answer on the esp32 forum and this place looks more active.

https://esp32.com/viewtopic.php?f=13&t=17738

I tried the blehr sample on a ESP32-WROOM and everything looks fine so far. But when I moved on a ESP32-CAM device the Android app connects and then disconnects right away.

I am attaching the logs of both cases. I tried to check them side by side but I understand too little to make anything out of it.

What difference between the two modules could be causing this issue? 

[ESP32-CAM.txt](https://github.com/espressif/esp-nimble/files/5395184/ESP32-CAM.txt)
[ESP32-WROOM.txt](https://github.com/espressif/esp-nimble/files/5395185/ESP32-WROOM.txt)

prasad-alatkar, 2020-10-19:
Hi @ffrige Thank you for posting your issue here and apologies for not responding on esp-forum, I get notified for issues and PR posted here so it gets easier to track the issues. 

The error code 531 suggests the remote user has terminated the connection. The error codes for NimBLE can be referred from here: [NimBLE return codes](http://mynewt.apache.org/master/network/ble_hs/ble_hs_return_codes.html ). From the logs it looks like the remote device (Android) has requested LTK and the ESP does not have the LTK so it returns the negative status to remote device, because of this remote terminates connection.

```
LE LTK Req. handle=0 rand=00 encdiv=0
looking up our sec; peer_addr_type=1 peer_addr=0x3a 0xd2 0x6b 0x3d 0x80 0x56  ediv=0x00 rand=0x0
ble_hs_hci_cmd_send: ogf=0x08 ocf=0x001b len=2
``` 
Have you modified `blehr` code ? If modified is it possible to share the code ? Can you also please try to delete the ESP device from paired-devices (or Bonded) section of Android phone ? Can you also please clear cache memory of the BLE app which you are using to connect to ESP ? Please let me know if any of the suggestions help you resolve the issue.

ffrige, 2020-10-19:
@prasad-alatkar Thank you for replying so fast!!

I did not modify anything in the blehr code, except for the device name string.
I tried to delete the app cache and remove the pairing on the Android device, but that did not help.

However, you pointing out to the LTK made me think it could be something related to security. So I went through the NimBLE parameters in menuconfig and by simple trial-and-error found the one that made it working. It's "Security manager secure connections (4.2)".

No idea what that means, but by disabling it, my application was able to connect correctly to the ESP32-CAM and read/write characteristics. Maybe the CAM is older than the WROOM and is missing some update?

In any case it works now and that's great. Thank you so much for your help, you're awesome! 

[image]

prasad-alatkar, 2020-10-19:
Ohhh, glad that your issue is resolved now. However, you should not be facing the issue even with Secure Connection enabled. If you are using `bleprph` app then I can think of a culprit here: [bleprph#360](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/bleprph/main/main.c#L360) . You can see in the example that key distribution gets disabled when Secure Connection is enabled. Not sure if it helps, however you may try to shift one of the `#endif` (from line no. 367/368) to line no. 364, idea is `ble_hs_cfg.sm_our_key_dist` or `ble_hs_cfg.sm_their_key_dist` should not be masked for Secure Connection enabled. Please let me know if it helps.

> No idea what that means, but by disabling it, my application was able to connect correctly to the ESP32-CAM and read/write characteristics. Maybe the CAM is older than the WROOM and is missing some update?  

If the esp-idf versions are same, I don't think there should be any difference in code behavior as far as NimBLE is concerned.

ffrige, 2020-10-20:
Hello! I tried the `bleprph` this morning and it actually worked fine without changing anything. As I said, I understand too little of BLE to go through the code and modify things. Besides, the `bleprph` is significantly more complex than the `blehr`. I wish these examples had tutorials to break down the code and explain what is being done in details. I feel like I'm hacking and reverse-engineering things here, not really the best way to learn a new system...

closed 2020-10-20 by ffrige

prasad-alatkar, 2020-10-20:
Hi @ffrige I can understand that `blehr` example is quite easy to go through than `bleprph`. Just to mention, `bleprph` example is modified in such a way that it also demonstrates various security methods in NimBLE. The examples here are mostly taken from upstream NimBLE and upstream NimBLE has nice breakdown of code in this tutorial: [bleprph-upstream](https://mynewt.apache.org/latest/tutorials/ble/bleprph/bleprph.html#services-characteristics-descriptors). This can help you ramp up quite faster onto NimBLE, I understand that IDF's `bleprph` is little different from upstream but once you go through the above tutorial it will be rather a easy walk to go through IDF's `bleprph`. Having said this, we will certainly look to provide better(one-stop) documentation for IDF's NimBLE examples.

ffrige, 2020-10-20:
Thank you, I will try to dig deeper into NimBLE when I find the time. For now, I was looking for a quick way to read and write some data, and the `blehr` was good enough. I don't really need any security feature.
 Unfortunately, while BLE is now working well, it caused the video stream on the WiFi channel to stop. I thought the two could  work well side by side, maybe I am missing some configuration parameter or I am not handling the communication functions optimally.

djavorek, 2023-01-29:
Bumping this issue. I experienced the same.
Unfortunately, I cannot open menuconfig with PlatformIO and see config option for "Security Manager secure connections".

```
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_CONNECT 
BLE client has connected.
D NimBLEServerCallbacks: onConnect(): Default
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_DISCONNECT
CRIT NimBLEServer: Disconnect - host reset, rc=531
D NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0
D NimBLEAdvertising: << Advertising start
BLE client has disconnected.
D NimBLEServerCallbacks: onDisconnect(): Default
D NimBLEAdvertising: >> Advertising start: customAdvData: 0, customScanResponseData: 0
W NimBLEAdvertising: Advertising already active
```