espressif/esp-nimble #20: Is it correct that there are paths that are not unlocked in ble_gap_ext_connect?

KAMEDAkyosuke, 2020-11-27:
https://github.com/espressif/esp-nimble/blob/f8f02740acdf4d302d5c2f91ee2e34444d405671/nimble/host/src/ble_gap.c#L3542-L3573

prasad-alatkar, 2020-11-27:
Hi @KAMEDAkyosuke , if I have understood you correctly, I believe you are having issue with using extended connect API (`ble_gap_ext_connect`) on ESP32. We need to enable 'Extended Advertising' (`BLE_EXT_ADV`) feature for this API to work and ESP32 hardware does not have support for this feature. You can refer: https://www.espressif.com/en/news/BLE_5.0_Certification where it is mentioned that _"Due to hardware limitations, ESP32 does not support such functions of Bluetooth LE 5.0 as 2M PHY, LE Long Range, and ADV Extensions"_

KAMEDAkyosuke, 2020-11-27:
@prasad-alatkar thank you for your replay😀
I didn't understand about the hardware limitations.

closed 2020-11-27 by KAMEDAkyosuke