espressif/esp-nimble #22: Device type is Classic and LE not LE only

iliapenev, 2021-01-30:
On ESP32-D0WDQ6 device `bleprph` example is running. 
When scan is performed with `NRF Connect` for Android, device type of ESP32 is shown as `CLASSIC and LE`.

This leads to problems with Android, where device appears as `Dual`. It is recognized right, but using `BluetoothGatt.connectGatt()` requires to be specified transport parameter, which should be `BluetoothDevice.TRANSPORT_LE` [link](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int))
Otherwise on connection, device disconnects with following error. 

There is also problem with Python Bleak lib and maybe other BLE tools.

Android build for SDK min version 24 (7.0)
`... device.connectGatt(this, false, gattCallback); ...`

Android logcat
```
D/BluetoothGatt: connect() - device: M:A:C, auto: false
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=11 device=M:A:C
E/DeviceActivity: Disconnected from GATT server.
```
How could be changed device type to be `LE only`?

Tested with esp-idf v4.2 and Android 9 device

dhrishi, 2021-02-19:
Hi @iliapenev  Sorry for the late reply. The bleprph example in v4.2 shows device type as "LE only" in nRFConnect for me. You can see the code here as well: https://github.com/espressif/esp-idf/blob/release/v4.2/examples/bluetooth/nimble/bleprph/main/main.c#L88-L93 that justifies the same.

Can you please check and confirm?

iliapenev, 2021-02-25:
Hello, I've checked on new dev boards and device type is "LE only". 

On board that shows dual type, I've done again full erase and everything is fine now. 

Probably something was left in NVS during development 

I'll close issue for now. Thanks!

closed 2021-02-25 by iliapenev

TheGuardianWolf, 2024-10-22:
Hey @iliapenev, I know this post is 3 years old now but I am having the same issue. How did you do the full erase?

rahult-github, 2024-10-22:
@TheGuardianWolf  , you can do  " idf.py erase-flash"

TheGuardianWolf, 2024-10-23:
Unfortunately I'd already tried that, no luck

I switched to a new chip and it is working perfectly