espressif/esp-nimble #26: Connect without scanning

Murtaza7745, 2021-05-01:
Hello there,
I am new to BLE and would like some help connecting ESP32 as a BLE client to a known BLE server address without scanning.
I am stuck and need some help.
For now, I am using the NIMBLE_CLIENT example, which too much time to find the server if surrounded by many other ble servers.

PLEASE HELP.

prasad-alatkar, 2021-05-05:
Hi @Murtaza7745 , as per my understanding, scanning is necessary before connection. What you can do is add `filter_policy = BLE_HCI_SCAN_FILT_USE_WL` to scan only required devices. You can also specify `duration_ms` in discovery process to limit time spent during discovery process. Please do remember, you need to add the required device to list/whitelist before enabling filter policy to use whitelist. You can use API: `ble_gap_wl_set` to add device to whitelist. Please let me know if it helps you in some way.

Murtaza7745, 2021-05-05:
thanks alot will try this

RahulEspressif, 2021-06-18:
> Hello there,
> I am new to BLE and using Arduino IDE and would like some help connecting ESP32 as a BLE client to a known BLE server address without scanning.
> I am stuck and need some help.
> For now, I am using the NIMBLE_CLIENT example, which too much time to find the server if surrounded by many other ble servers.
> 
> PLEASE HELP.

Hi @Murtaza7745 ,   There is a way , where you can try to connect to remote, without the need to scan. Once client scans, it becomes aware of remote device details, like BD address.  If you already know the remote's bd address, you can try calling ble_gap_connect API directly from your code and pass the correct parameters ( remote device address etc).  If remote is advertising, then it should go for connection.

rahult-github, 2023-01-11:
Closing this ticket due to no activity.. Please feel free to reopen if issue is still not resolved