espressif/esp-nimble #27: Nimble stack resets with reason 0x13

Teu97, 2021-06-10:
Hello again,

I'm implementing the nimble Bluetooth stack (esp-idf v4.1.1) into my ESP32-WROOM-32D project which has been set as BLE peripheral. When I walk away with my Phone or disable bluetooth, and then come back in range or enable bleutooth and try to connect again, the nimble reset_cb is called with error 0x13 (BLE_HS_ETIMEOUT_HCI).

I've got two devices which are currently bonded, 1 still connected, on the other I execute the steps as described above. The problem is that I cannot always reproduce the issue, my guess is that this is due to Bluetooth addresses changing is not always at the same moment?

My bluetooth device has been set up as followed:
```C++
void CEspNimbleBluetoothDevice::Init()
{
  esp_err_t result;
  result = esp_nimble_hci_and_controller_init();
  Assert(result == ESP_OK);

  nimble_port_init();

  ble_hs_cfg.reset_cb = PeripheralOnResetCallback;
  ble_hs_cfg.sync_cb = PeripheralOnSyncCallback;
  ble_hs_cfg.gatts_register_cb = GattRegisterEventCallback;
  ble_hs_cfg.gatts_register_arg = reinterpret_cast<void*>(this);
  ble_hs_cfg.store_status_cb = ble_store_util_status_rr;

  ble_hs_cfg.sm_io_cap = BLE_SM_IO_CAP_NO_IO; // No input/output capabilities (just works)
  ble_hs_cfg.sm_bonding = 1; // Enable bonding
  ble_hs_cfg.sm_sc = 1; // Use secure connection
  ble_hs_cfg.sm_our_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID; // Enable LTK + IRK
  ble_hs_cfg.sm_their_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID; // Enable LTK + IRK

  ble_svc_gap_init();
  ble_svc_gatt_init();

  memset(m_serviceTable, 0, sizeof(m_serviceTable));
  memset(m_characteristicRegistrationTable, 0, sizeof(m_characteristicRegistrationTable));
}
```
See SDK config for more information: [sdkconfig.txt](https://github.com/espressif/esp-nimble/files/6632082/sdkconfig.txt)

The image below contains my debug log:
[image]

The following things in my log stand out:
- Unhandled event 0x10 - BLE_GAP_EVENT_ENC_CHANGE
- Unhandled event 0x15 - BLE_GAP_EVENT_MTU

_Looking at the bleprph example both events are not required to be handled but will be returned with 0 directly._ 

- The reset_cb is called, with error: 0x13 - BLE_HS_ETIMEOUT_HCI
- Advertising start error: 0x15 - BLE_HS_ENOADDR, probably because the nimble stack just reset, see error 0x13

Hopefully somebody can help me with this.

Teu97, 2021-07-15:
Hi all, I'm still having trouble with this issue. Can anyone help?

rahult-github, 2025-07-29:
This is an old issue and various nimble version changes have gone in ESP-IDF. Closing issue. Feel free to reopen if issue seen with any latest releases.

closed 2025-07-29 by rahult-github