espressif/esp-nimble #28: NimBLE: Unable to find the second device of the same product after using Host based privacy (RPA) feature support (v4.0)

soonyak, 2021-07-01:
**Environment**

Module or chip used: ESP32-WROOM-32D
IDF version: v4.0 beta2 (commit @99fb9a3f7)
NimBLE version: v1.1.0 (commit @a5127ec)

**Problem Description**

I have ported the Host based privacy (RPA) feature support (v4.0) to my code and this is able to the connection problem when iPhone/Android devices use RPA addresses, which change periodically or after Bluetooth on-off. However, when the phone successfully pairs the first device, it is not able to find the second device of the same product.

Is this the normal behaviour? I am not able to find any description about this online.

I have attached my init code sync callback function that implements the following:

 1. Set own_addr_type to `BLE_OWN_ADDR_RANDOM`.
 2. Add `BLE_SM_PAIR_KEY_DIST_ID` to key distribution in
    `ble_hs_cfg.sm_our_key_dist` & `ble_hs_cfg.sm_their_key_dist`.
 3. Call `ble_hs_pvcy_rpa_config(1)` in Host-Controller sync callback.

[AppCode.txt](https://github.com/espressif/esp-nimble/files/6745153/AppCode.txt)

soonyak, 2021-07-01:
I can only see the second device only when I unpair the first device.

rahult-github, 2024-02-08:
This is old issue and also reported on older IDF version which is EOL. Please let us know if issue still observed on latest releases. Closing the issue.

closed 2024-02-08 by rahult-github