espressif/esp-nimble #29: Cannot compile due to type difference (error: invalid conversion from 'int' to 'esp_log_level_t' )

hogesako, 2021-07-11:
# Enviroment
PLATFORM: Espressif 32 (3.3.0) > Espressif ESP32 Dev Module
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)

# Problem Description
The following compile error occurs when using MODLOG_DFLT.
```
/Users/hogesako/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/include/log_common/log_common.h:30:29: error: invalid conversion from 'int' to 'esp_log_level_t' [-fpermissive]
 #define LOG_LEVEL_INFO     (1)
                            ~^~
/Users/hogesako/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/include/modlog/modlog.h:38:60: note: in definition of macro 'MODLOG_ESP_LOCAL'
     if (MYNEWT_VAL(BLE_HS_LOG_LVL) <= level) esp_log_write(level, "NimBLE", ml_msg_, ##__VA_ARGS__); \
                                                            ^~~~~
/Users/hogesako/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/include/modlog/modlog.h:45:22: note: in expansion of macro 'LOG_LEVEL_INFO'
     MODLOG_ESP_LOCAL(LOG_LEVEL_INFO, ml_msg_, ##__VA_ARGS__)
                      ^~~~~~~~~~~~~~
/Users/hogesako/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/include/modlog/modlog.h:96:5: note: in expansion of macro 'MODLOG_INFO'
     MODLOG_ ## ml_lvl_((ml_mod_), __VA_ARGS__)
     ^~~~~~~
/Users/hogesako/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/porting/nimble/include/modlog/modlog.h:99:5: note: in expansion of macro 'MODLOG'
     MODLOG(ml_lvl_, LOG_MODULE_DEFAULT, __VA_ARGS__)
```

This error appears to be the result of the following change.
https://github.com/espressif/esp-nimble/commit/f0ad8684b86660dde21de5cd9069ea12021b82c1#diff-5ccf5455a3cb833edf887e0cfa93e7d1a93a1f46852d9df30174e0633265e34f

RahulEspressif, 2021-07-16:
Hi @hogesako ,  Thanks for reporting. Can you please confirm if attached patch works for you ? 
[error_conversion_macro.txt](https://github.com/espressif/esp-nimble/files/6827922/error_conversion_macro.txt)

hogesako, 2021-07-17:
Hi @RahulEspressif ,Thanks for replying.
I applied your patch and the compile worked.

closed 2022-03-23 by rahult-github