espressif/esp-nimble #30: esp_log_level_set("NimBLE") does not work in 4.3

jhnlmn, 2021-07-20:
esp-idf-v4.3/components/bt/host/nimble/nimble/porting/nimble/include/log_common/log_common.h
has inverted priorities:
#define LOG_LEVEL_DEBUG    (0)
#define LOG_LEVEL_INFO     (1)
#define LOG_LEVEL_WARN     (2)
#define LOG_LEVEL_ERROR    (3)
#define LOG_LEVEL_CRITICAL (4)
#define LOG_LEVEL_NONE     (5)
/* Up to 7 custom log levels. */
#define LOG_LEVEL_MAX      (15)

vs regular
esp-idf-v4.3/components/log/include/esp_log.h
typedef enum {
    ESP_LOG_NONE,       /*!< No log output */
    ESP_LOG_ERROR,      /*!< Critical errors, software module can not recover on its own */
    ESP_LOG_WARN,       /*!< Error conditions from which recovery measures have been taken */
    ESP_LOG_INFO,       /*!< Information messages which describe normal flow of events */
    ESP_LOG_DEBUG,      /*!< Extra information which is not necessary for normal use (values, pointers, sizes, etc). */
    ESP_LOG_VERBOSE     /*!< Bigger chunks of debugging information, or frequent messages which can potentially flood the output. */
} esp_log_level_t;

As the result, 
esp_log_level_set("NimBLE", ESP_LOG_WARN);
used to work in 4.2, but does not work in 4.3.
Logs like "GATT procedure initiated" are still printed

RahulEspressif, 2021-07-21:
Hi @jhnlmn , Thanks for reporting this issue. We are already working internally on fixing this. Please confirm if attached change works for you. 
[error_conversion_macro.txt](https://github.com/espressif/esp-nimble/files/6852943/error_conversion_macro.txt)

jhnlmn, 2021-07-23:
> Please confirm if attached change works for you.
I do not know.
Currently I can only use official IDF releases.
For now I will use
CONFIG_BT_NIMBLE_LOG_LEVEL_WARNING=y
.

rahult-github, 2022-03-10:
Hi @jhnlmn ,

 Please check with latest release, where esp_log_level_set should work as expected . 

Thanks,
 Rahul

closed 2022-03-10 by rahult-github