espressif/esp-nimble #32: Whitelisting RPA devices with local Random Static Address.

happy-sloth, 2021-09-19:
Hello!

I am currently working on a device that should work like this: 
1. The device is invisible to all non-whitelisted devices (filter policy is 3) and have Static Random Address.
2. The process of pairing with a new Central device is started by pressing the button. Pressing the button switches the filtering policy to mode 0.
3. After pairing, the device is added to the whitelist, and the whitelist is saved in non-volatile memory so that when the device is rebooted it can be uploaded to the working whitelist. 

When I connect from my smartphone to the device, the pairing and bonding process is successful. The smartphone reconnects and accepts advertising packages, which means that the whitelist is working. But, unfortunately, when a smartphone changes its RPA address, the pairing process has to be repeated, because for ESP32 the smartphone has become a different device with a different address. 

I went through all the threads in this repository and none helped me.  

I use ESP-IDF v4.3.1, and nimble-1.3.0-idf-v4.3-afr.

In addition my main.c file and log-file.
[main.txt](https://github.com/espressif/esp-nimble/files/7192157/main.txt)
[nimble_periph_Log.txt](https://github.com/espressif/esp-nimble/files/7192158/nimble_periph_Log.txt)

happy-sloth, 2021-09-20:
if I change Key Distribution Mask to:

ble_hs_cfg.sm_our_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
ble_hs_cfg.sm_their_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;

then I can't roconnect to ESP32 until I clean the bond list on smartphone or reboot ESP32. But when a smartphone changes its RPA address ESP32 still can no longer recognize him by the whitelist.

happy-sloth, 2022-09-08:
Hi @rahult-github !

I was working on commercial projects and there was a very tight deadline, so I quickly switched to bluedroid and implemented a manual check of the address for presence in the bond list. And completely forgot about my question. Does it make sense to continue this? Because I already forgot what the problem was.

rahult-github, 2022-09-16:
May be we can close this and you can reopen a new one when you observe the issue .

closed 2022-09-16 by rahult-github