espressif/esp-nimble #33: iOS and Android bonding issues

flonas955, 2021-10-07:
Hello,

I am currently developing an app for ESP32 using NimBLE, that will pair and bond with an Android and iOS application. I am using "just works" pairing method. 

I have some problems with bonding, especially on iOS, where I receive the _Error: Peer removed pairing information_ message. It will only pair again if I forget the device from the Bluetooth settings. I know that iOS uses RPA addresses and I have configured the board to resolve RPA addresses as detailed [here](https://www.esp32.com/viewtopic.php?t=14966#p58984), but it does not seem to work. Every time the iPhone changes its address, the bonding is no longer detected and the error appears.

With random addresses disabled on ESP32, there are problems with bonding persistence on Android also, but it appears only occasionally on Android devices. With random addresses enabled, the connection will no longer work on this OS.

I have already posted related questions on the ESP-IDF forum, but I had no aswers ([here](https://www.esp32.com/viewtopic.php?f=13&t=22753) and [here](https://www.esp32.com/viewtopic.php?f=13&t=23276))

The current nimble-idf branch is **nimble-1.3.0-idf** and the latest commit  hash is **5bb7b40**. I can provide any extra information you might need.

Any suggestion would be very helpful.

Thank you,
Florin

flonas955, 2021-11-08:
Another problem is that some phones are not able to detect the ESP32 with RPA resolution active, while advertising. The solution is to disable and then reenable the Bluetooth on the phone regularly. Any ideas on this regard?

SlightlyStoopid, 2022-11-28:
I'm not sure to your exact setup on how your handling your side. But I have been dealing with iOS bonding issues for awhile and can relate. So in general as a PSA because I see a lot of issues here relating to bonding/rpa and it took me awhile to get it going, I'll go over the jist of whats needed.

If you take a look at the bleprph example, that uses a store template to save the irk/ltk/csrk of the central and has the ESP in an NRPA configuration if random is set, for the ESP in RPA mode the "ble_hs_pvcy.h" details the steps for handling RPA on the ESP side. Using both the store template and the "ble_hs_pvcy.h" steps this will handle RPA-RPA connections well.

So I would check that you're setting the store template as described in the bleprph example as well as following the steps in the ble_hs_pvcy if you are setting your own device to RPA.

In general for RPA-RPA

I would setup your sync as follows: 

```    
    ble_hs_pvcy_rpa_config(true);
    rc = ble_hs_util_ensure_addr(1);
    assert(rc == 0);

    /* Figure out address to use while advertising (no privacy for now) */
    rc = ble_hs_id_infer_auto(1, &own_addr_type);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error determining address type; rc=%d\n", rc);
        return;
    }
```
*Side note I've seen reference [here](https://github.com/espressif/esp-nimble/issues/8) that _infer_auto is unnecesary but this works for me so idk.

Configure your ble_hs_config ensuring it has the store and key dists if you're in RPA mode on ESP
```
    ble_hs_cfg.store_status_cb = ble_store_util_status_rr;
    ble_hs_cfg.sm_io_cap = BLE_SM_IO_CAP_DISP_ONLY;
    ble_hs_cfg.sm_bonding = 1;
    ble_hs_cfg.sm_mitm = 1;
    ble_hs_cfg.sm_sc = 1;
    ble_hs_cfg.sm_our_key_dist = 1 | BLE_SM_PAIR_KEY_DIST_ID;
    ble_hs_cfg.sm_their_key_dist = 1 | BLE_SM_PAIR_KEY_DIST_ID;
```

And define and call your template as described in the example

```
    /* Define template prototype for store*/
    void ble_store_config_init(void);

    /* XXX Need to have template for store */
    ble_store_config_init();
```

I hope this helps in anyway, just for reference this also worked for me on release/v5.0 of esp-idf

Best Regards,
Stoop

veneno-529, 2022-12-19:
Hello, I have done the same procedure as you have mentioned keeping, own_addr_type = BLE_ADDR_RANDOM.
I have also added store template ble_store_config_init(); but the function is empty do we have write some code in it to store the IRK keys?.
During debugging i found that RPA-NRPA error message was printed and it said to scan and connect again.

veneno-529, 2023-01-08:
> I'm not sure to your exact setup on how your handling your side. But I have been dealing with iOS bonding issues for awhile and can relate. So in general as a PSA because I see a lot of issues here relating to bonding/rpa and it took me awhile to get it going, I'll go over the jist of whats needed.
> 
> If you take a look at the bleprph example, that uses a store template to save the irk/ltk/csrk of the connection and has the ESP in an NRPA configuration if random is set, for the ESP in RPA mode the "ble_hs_pvcy.h" details the steps for handling RPA on the ESP side. Using both the store template and the "ble_hs_pvcy.h" steps this will handle RPA-RPA connections well.
> 
> So I would check that you're setting the store template as described in the bleprph example as well as following the steps in the ble_hs_pvcy if you are setting your own device to RPA.
> 
> In general for RPA-RPA
> 
> I would setup your sync as follows:
> 
> ```
>     ble_hs_pvcy_rpa_config(true);
>     rc = ble_hs_util_ensure_addr(1);
>     assert(rc == 0);
> 
>     /* Figure out address to use while advertising (no privacy for now) */
>     rc = ble_hs_id_infer_auto(1, &own_addr_type);
>     if (rc != 0) {
>         MODLOG_DFLT(ERROR, "error determining address type; rc=%d\n", rc);
>         return;
>     }
> ```
> 
> *Side note I've seen reference [here](https://github.com/espressif/esp-nimble/issues/8) that _infer_auto is unnecesary but this works for me so idk.
> 
> Configure your ble_hs_config ensuring it has the store and key dists if you're in RPA mode on ESP
> 
> ```
>     ble_hs_cfg.store_status_cb = ble_store_util_status_rr;
>     ble_hs_cfg.sm_io_cap = <Your IO Cap>;
>     ble_hs_cfg.sm_bonding = 1;
>     ble_hs_cfg.sm_mitm = 1;
>     ble_hs_cfg.sm_sc = 1;
>     ble_hs_cfg.sm_our_key_dist = 1 | BLE_SM_PAIR_KEY_DIST_ID;
>     ble_hs_cfg.sm_their_key_dist = 1 | BLE_SM_PAIR_KEY_DIST_ID;
> ```
> 
> And define and call your template as described in the example
> 
> ```
>     /* Define template prototype for store*/
>     void ble_store_config_init(void);
> 
>     /* XXX Need to have template for store */
>     ble_store_config_init();
> ```
> 
> I hope this helps in anyway, just for reference this also worked for me on release/v5.0 of esp-idf
> 
> Side note for anyone interested in setting/configuring their own IRK/LTK/CSRK's and not using the standard default credit [here](https://github.com/apache/mynewt-nimble/pull/1396) This ports cleanly into esp-nimble, though it does require overriding the bt component which may not be desired
> 
> Best Regards, Stoop

Please let me know what should be own_addr_type ? BLE_OWN_ADDR_RANDOM or BLE_ADDR_RANDOM.
I've tried to use your method and I'm noticing that peer records are not getting stored as per my logs. Device is getting added in resolved list but records are not being stored. I have erased flash and tried to program it again still failure.

veneno-529, 2024-05-09:
Hello @flonas955 @SlightlyStoopid 

I am using ESP-IDF 5.1.1 and nimble. I have enabled host based privacy for Random address. Despite following all the steps mentioned in the thread. I'm still facing the same issue. 

I have configured nimble for below settings as follows -

Maximum number of concurrent connects = 1
Maximum number of bonds to save across boots = 1
Maximum number of CCC descriptors to save across reboots = 12
Host based privacy for random address = enabled
BLE MAX connections = 3

`/* Initialize the NimBLE host configuration. */
    ble_hs_cfg.reset_cb = ble_app_on_reset;
    ble_hs_cfg.sync_cb = ble_app_on_sync;
    ble_hs_cfg.gatts_register_cb = gatt_svr_register_cb;
    ble_hs_cfg.store_status_cb = ble_store_util_status_rr;

    ble_hs_cfg.sm_io_cap         = BLE_SM_IO_CAP_DISP_ONLY;
    ble_hs_cfg.sm_bonding        = 1;
    ble_hs_cfg.sm_sc             = 1;
    ble_hs_cfg.sm_our_key_dist   = 1;
    ble_hs_cfg.sm_their_key_dist = 1;
    ble_hs_cfg.sm_mitm           = 1;
    ble_hs_cfg.sm_our_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
    ble_hs_cfg.sm_their_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
    
    ble_svc_gap_device_name_set(device_id);           // 1 - Initialize NimBLE configuration - server name
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_N12);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_N12);
    esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN ,ESP_PWR_LVL_N12);
    ble_svc_gap_init();                        // 2 - Initialize NimBLE configuration - gap service
    ble_svc_gatt_init();                       // 3 - Initialize NimBLE configuration - gatt service
    ble_svc_ans_init();                        
    ble_gatts_count_cfg(gatt_svcs);            // 4 - Initialize NimBLE configuration - config gatt services
    ble_gatts_add_svcs(gatt_svcs);             // 5 - Initialize NimBLE configuration - queues gatt services.

    /* XXX Need to have template for store */
    ble_store_config_init();`

`void ble_app_on_sync(void)
{
    int rc;
    ble_hs_pvcy_rpa_config(true);
    rc = ble_hs_util_ensure_addr(1);
    assert(rc == 0);
    /* Figure out address to use while advertising (no privacy for now) */
    rc = ble_hs_id_infer_auto(1, &ble_addr_type);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error determining address type; rc=%d\n", rc);
        return;
    }
    /* Printing ADDR */
    uint8_t addr_val[6] = {0};
    rc = ble_hs_id_copy_addr(ble_addr_type, addr_val, NULL);

    MODLOG_DFLT(INFO, "Device Address: ");
    print_addr(addr_val);
    MODLOG_DFLT(INFO, "\n");
    /* Begin advertising by pressing pairing button */
}`

Please help me out.

rahult-github, 2024-05-09:
Hi @veneno-529   , can you please share the chip being used AND please help share debug enabled logs .

veneno-529, 2024-07-09:
Hi @rahult-github any update regarding the issue?

serdartoren, 2025-04-21:
By the way, when I use the "Just_work" method, there is no problem. If I try to connect to safely, there will ocur this problem.

rahult-github, 2026-08-18:
This is an old issue on branch which are EOL. Request to check any latest active branch and reopen in case issue is still observed.

closed 2026-08-18 by rahult-github