espressif/esp-nimble #37: Bonding problem with ESP32-C3

stevegrau-ayla, 2021-11-16:
With IDF v4.3.1 bonding doesn't work on the ESP32-C3 while it does work with the same code running on the ESP32. I've determined the that it isn't working because on the ESP32 the BLE_SM_OP_IDENTITY_INFO and BLE_SM_OP_IDENTITY_ADDR_INFO events are received by the host task while it is in state BLE_SM_PROC_STATE_KEY_EXCH while on the ESP32-C3 these events arrive while in the state BLE_SM_PROC_STATE_ENC_START. Changing the state prerequisite argument in the calls to ble_sm_proc_find to BLE_SM_PROC_STATE_NONE in the calls from ble_sm_id_info_rx and ble_sm_id_addr_info_rx enables bonding to succeed on ESP32-C3 as it does on the ESP32.

The app I'm working on uses simple no-passkey pairing and the device is acting as a peripheral.

strange969, 2025-09-10:
Hi @stevegrau-ayla , sorry for the delayed response. Could you please try this on the latest tag and let us know if the issue still occurs?

rahult-github, 2026-08-18:
Closing due to no activity. Feel free to reopen in case of further updates.

closed 2026-08-18 by rahult-github