espressif/esp-nimble #38: memory leaking in provisioning procedure for mesh provisioner

justabetterman, 2021-11-26:
I found a memory leaking issue caused by mbedtls_gen_keypair as below, i think mbedtls_ecp_keypair_free should be invoked regardless of the rc vaule.

exit:
    mbedtls_ctr_drbg_free( &ctr_drbg );
    mbedtls_entropy_free( &entropy );
    if (rc != 0) {
        mbedtls_ecp_keypair_free(&keypair);
        return BLE_HS_EUNKNOWN;
    }

the memory leaking issue can be observed when device works as a mesh provisioner and finish provisioning node device many times.

rahult-github, 2022-05-30:
keypair is a static variable which gets inited everytime. There should be a check to free the keypair (if it is already allocated) . Will add this check.

rahult-github, 2022-09-16:
This [commit](https://github.com/espressif/esp-nimble/commit/d48bc4e78ef2478a0ac32db3d832d1e8178451db) address the issue.  Sorry for delay in update.

closed 2022-09-16 by rahult-github