espressif/esp-nimble #40: How to enable Data Length Extension (DLE)?

HarryWindsor, 2021-12-08:
One of the features highlights of NimBLE is the "support for 251 byte packet size". 
How do I enable "Data Length Extension" (DLE) to send packages of size 251 instead of 27 bytes of data?

I have set my MTU to 256 bytes and just send dummy data. This generates multiple messages of size 32 bytes (each with 27 bytes of data).

Below is a screenshot from wireshark:
[image]

rahult-github, 2022-01-07:
Hi, 
 ble_hs_hci_util_set_data_len is the API available [here](ble_hs_hci_util_set_data_len(uint16_t conn_handle, uint16_t tx_octets,) that can be used for this. 

Thanks,
 Rahul

MassimilianoMontagni98, 2022-01-07:
Thanks for the response but the function signature that I have is slightly different: 

`int ble_hs_hci_util_set_data_len(uint16_t conn_handle, uint16_t tx_octets,  uint16_t tx_time)`

how about the tx_time parameter?

Thanks,
Massimiliano

rahult-github, 2022-01-07:
My bad, copy pasted incorrect line. The signature you mentioned is correct: 

https://github.com/espressif/esp-nimble/blob/nimble-1.3.0-idf/nimble/host/src/ble_hs_hci_util.c#L125

MassimilianoMontagni98, 2022-02-02:
@rahult-github sorry for the late response but I had corona, I tried as you suggested `ble_hs_hci_util_set_data_len` and the return code is `0` but when I try to transfer a file is still sent in small packets

HarryWindsor, 2022-10-02:
It has be enabled from the client

closed 2022-10-02 by HarryWindsor