espressif/esp-nimble #42: Bug: ".../nimble/host/include/host/ble_uuid.h:97:9: error: expected primary-expression before '.' token"

Kampi, 2022-05-13:
## Environment

- IDF version (run ``git describe --tags`` to find it): 4.4.1

## Problem Description

Using `static const ble_uuid128_t gatt_svr_svc_sec_test_uuid = BLE_UUID128_INIT(0x2d, 0x71, 0xa2, 0x59, 0xb4, 0x58, 0xc8, 0x12, 0x99, 0x99, 0x43, 0x95, 0x12, 0x2f, 0x46, 0x59);` from the NimBLE `bleprph` example will produce the error:

```
In file included from C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs_adv.h:24,
                 from C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_gap.h:32,
                 from C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:33,
                 from src/Bluetooth/bluetooth.cpp:9:
C:/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/include/host/ble_uuid.h:97:9: error: expected primary-expression before '.' token
         .u.type = BLE_UUID_TYPE_128,    \
         ^
```

The problem can be solved by fixing the macro in `ble_uuid.h`:

```
#define BLE_UUID128_INIT(uuid128...)    \
    {                                   \
        .u = {                          \
            .type = BLE_UUID_TYPE_128,  \
        },                              \
        .value = { uuid128 },           \
    }
```

rahult-github, 2022-05-13:
Hi @Kampi ,  

 Thanks for the changes. As we use upstream-nimble code and try to be in sync, I wonder if you can also create a  PR on [mynewt-nimble repo ](https://github.com/apache/mynewt-nimble) ?  I think , this change should go not just in ESP-IDF but even in upstream nimble

Thanks,
 Rahul

Kampi, 2022-05-13:
Hi @rahult-github 

I have created a pull request in the `mynewt-nimble` repository ([Link](https://github.com/apache/mynewt-nimble/pull/1259)).

Kampi, 2022-05-27:
The fix was merged into the nimble repo.

closed 2022-05-27 by Kampi