espressif/esp-nimble #49: LOG_LEVEL defines collision

vovagorodok, 2022-11-30:
At `log_common.h` we have defined:
```
#define LOG_LEVEL_DEBUG    (0)
#define LOG_LEVEL_INFO     (1)
#define LOG_LEVEL_WARN     (2)
#define LOG_LEVEL_ERROR    (3)
#define LOG_LEVEL_CRITICAL (4)
#define LOG_LEVEL_NONE     (5)
```
`LOG_LEVEL..` names are very common and I have redefinition issues when another than NibBle logger is used.\
Because its internal NibBle logger, proposed to use some internal names like:
`BLE_LOG_LEVEL..` or `NIM_BLE_LOG_LEVEL..`

rahult-github, 2022-12-01:
Hi @vovagorodok , 

ESP-IDF clones upstream mynewt-nimble and we try to be in sync with upstream code. This file is actually coming from upstream nimble code. i suggest you to also open a ticket [here](https://github.com/apache/mynewt-nimble) .

Thanks,
 Rahul

vovagorodok, 2022-12-01:
Done, lets wait for response: https://github.com/apache/mynewt-nimble/issues/1416

vovagorodok, 2023-10-06:
Hi, done from apache side. Lets align esp-nimble repo now :)

vovagorodok, 2023-11-22:
@rahult-github what plans on this topic?

Staars, 2023-12-31:
I have the exact same problems with colliding defines and would like to this solved.

vovagorodok, 2024-03-17:
@rahult-github there is some issue with merging upstream? Can I help somehow?

rahult-github, 2024-03-18:
Hi @vovagorodok , sorry for delay. 

Current nimble version of IDF is 1.5.0 .. Backporting the patch to this branch has more changes than anticipated. We are in process of migrating to IDF 1.6.0. So the patch will be applied once we migrate. Also, we will backport it for nimble-1.5.0-idf too.  Will update here soon for this. 

In meantime, i will rework on the backporting of the patch and share it here for your usage, that will help unblock you untill it makes way to official IDF github code

vovagorodok, 2024-11-16:
@rahult-github what is the current situation? As I see porting to IDF 1.6.0 is done?

rahult-github, 2026-08-18:
Nimble version is bumped to 1.9.0-idf . Closing issue

closed 2026-08-18 by rahult-github