espressif/esp-nimble #54: Pairing process passkey method

suadsuljic7, 2023-03-28:
Hi I'm trying to implement the pairing process (passkey method) using Nimble Ble stack on two esp32. One device will initiate the pairing process([b]ble_gap_security_initiate[/b]) the other device needs to send the pin (pin is hard coded).
So I used the bleprph example with minor changes, and using my phone everything is working fine! When trying to connect I need to set the pin code...

But I need this part on a second esp32 I used the blecent example for the second device, when the ble_gap_security_initiate function is called from the central device shouldn't be there a GAP EVENT on the second device? (BLE_GAP_EVENT_PASSKEY_ACTION). How I know the pairing process has started?

rahult-github, 2023-03-28:
The existing applications in IDF are taken as is from upstream nimble.

bleprph example has been modified to showcase how to set io capability and other fields. 

On blecent side, since the ble_hs_cfg structure is global, so if the application is used as is, then the values for fields like sm_io_cap, bonding / mitm / sc are all by default set to 0. If the requirement is to enhance the blecent application to catch the events,  then changes done in bleprph should be referred and same changes should be done even on blecent side.  Then, the required events should be received.

suadsuljic7, 2023-03-29:
Hi,  on the blecent I already changed the ble_hs_cfg.  The changes I made on the central are only when a BLE_GAP_EVENT_CONNECT is received and the Connection is established, I call the ble_gap_security_initiate function, after that I immediately receive the BLE_GAP_EVENT_ENC_CHANGE on both sides, I cannot get the event BLE_GAP_EVENT_PASSKEY_ACTION. 

When the ble_hs_cfg is configured for the passkey method and when calling ble_gap_security_initiate on the central device, shouldn't the peripheral get the BLE_GAP_EVENT_PASSKEY_ACTION?

suadsuljic7, 2023-03-29:
Sure, It is the blecent example with minor changes in main.c https://drive.google.com/file/d/1QQWeEWHczhC1iNvoAdyLHdhFya46w2P1/view?usp=share_link
Also the bleperh is the same just changed the ble_hs_cfg.sm_io_cap and it is working with a phone. I can also upload it if need it. Thanks

rahult-github, 2023-03-29:
Hi @suadsuljic7 ,

I took only the below code from your blecent /main.c  and used bleprph from master branch.  Attaching sdkconfig for both bleprph and blecent .   Since default IOCAP is JustWorks for bleprph , i modified it to use KeyboardDisplay for testing purpose. 

With these changes, on bleprph side: 
----
I (13353) NimBLE_BLE_PRPH: PASSKEY_ACTION_EVENT started 

I (13353) NimBLE_BLE_PRPH: Enter the passkey through console in this format-> key 123456
----

on blecent, i added acase for PASSKEY entry ( change shared ) and few extra prints and see it is getting passkey action event. 

---
I (1015) NimBLE: Connection established 
I (1015) NimBLE: 

Starting SECURITY !! 
I (1025) NimBLE: Connection secured

Got Passkey Action 
---

You can have a look at attached changes and change application.  

The main.c.txt attached here is the blecent/main/main.c file that i modified at my end to take your changes partially. 

[bleprph_sdkconfig.txt](https://github.com/espressif/esp-nimble/files/11101691/bleprph_sdkconfig.txt)
[blecent_sdkconfig.txt](https://github.com/espressif/esp-nimble/files/11101692/blecent_sdkconfig.txt)
[main.c.txt](https://github.com/espressif/esp-nimble/files/11101702/main.c.txt)

suadsuljic7, 2023-03-30:
Hi, @rahult-github 

I m not sure what I m doing wrong! So i pull the latest commit on master branch, in the bleprph using the menuconifg IOCAP I changed Keyboard and Display, sdkconfig looks the same(also tried to copy).  On the blecent side also from the master branch I copied your main.c file sdkconfig looks the same(also tried to copy). I m not sure what am I doing wrong. Here is the output from the serial the upper is the peripheral and below is central
[image]

rahult-github, 2023-03-30:
Hi @suadsuljic7  , do you observe any differences in the sdkconfig  i shared vs the ones your application has ? The NIMBLE specific and NIMBLE_EXAMPLE specific ?

suadsuljic7, 2023-03-30:
Not that I see, this are the only differences 
[image]

rahult-github, 2023-03-30:
Can you please also share the bleprph side sdkconfig difference  ?

suadsuljic7, 2023-03-30:
Sure!
[image]

rahult-github, 2023-03-30:
Hi @suadsuljic7  ,

Here is my suggestion.  Please use attached bleprph and blecent "as is ".  I am also attaching the expected console logs for both peripheral and central.  Please check if you observe the same behaviour of getting PASSKEY_ENTRY print on both central and peripheral side. 

If yes, then you can use this template and modify the blecent application as per your use case. 

Since i have ran the test for esp32 at my end, please ensure you use the same sdkconfig. 

[blecent_console.txt](https://github.com/espressif/esp-nimble/files/11112472/blecent_console.txt)
[bleprph_console.txt](https://github.com/espressif/esp-nimble/files/11112473/bleprph_console.txt)
[blecent.zip](https://github.com/espressif/esp-nimble/files/11112474/blecent.zip)
[bleprph.zip](https://github.com/espressif/esp-nimble/files/11112475/bleprph.zip)

I am on latest master branch for the above test.

suadsuljic7, 2023-03-30:
Hi @rahult-github Thank you!
It seems to work I m not sure what the problem was I need to check it maybe the sdkconfig. Im receiving the PASSKEY_ACTION_EVENT. Thanks again.

closed 2023-03-31 by rahult-github