espressif/esp-nimble #62: introducing struct ble_gatt_cpfd containing a member named 'namespace' causes issue when compiled using C++ compiler

MarleenDev, 2024-02-07:
In C++, namespace is a reserved keyword, and cannot be used to name variables.
In code introduced in commit 0273e0e83, a struct is introduces which has a member called namespace.
I noticed the problem when trying to compile the esp-matter light example. This uses the esp-idf repo, which recently updated the nimble submodule to include the commit 0273e0e83, resulting in the example no longer building.

MarleenDev, 2024-02-07:
struct was added to nimble/host/include/host/ble_gatt.h

rahult-github, 2024-02-07:
Hi @MarleenDev  , thanks for pointing this. Please try attached patch which should fix the issue. 
[0001-update-variable-name-to-avoid-using-keyword.txt](https://github.com/espressif/esp-nimble/files/14193195/0001-update-variable-name-to-avoid-using-keyword.txt)

rahult-github, 2024-03-12:
Changes merged, closing

closed 2024-03-12 by rahult-github