espressif/esp-nimble #64: Re-connect to peripheral device fails on IDF v4.4.7

andreaslundeen, 2024-03-07:
Works in 4.4.6. But the updates in https://github.com/espressif/esp-nimble/commit/c50690e1b7794b9485921b1991b86387141832c0 breaks this and I am unable to re-connect. 
Seems that the added lines in ble_gap_connect() should be guarded with :
#if MYNEWT_VAL(BLE_HOST_BASED_PRIVACY)
    if (ble_host_rpa_enabled())
    {
    struct ble_hs_resolv_entry *rl = NULL;
    rl = ble_hs_resolv_list_find(bhc_peer_addr.val);

    if (rl != NULL) {
        BLE_HS_LOG(INFO, "%08x", *(uint32_t*)rl->rl_peer_rpa);
        memcpy(bhc_peer_addr.val, rl->rl_peer_rpa, BLE_DEV_ADDR_LEN);
        bhc_peer_addr.type = rl->rl_addr_type;
    }
    }
#endif

rahult-github, 2024-03-07:
Hi @andreaslundeen  , Thanks for reporting, we will fix this asap.

darshan7patel, 2024-03-07:
Hi @andreaslundeen , Kindly use the attahced patch. Please let me know if any addtional information is required.
[reconnect_failure.txt](https://github.com/espressif/esp-nimble/files/14522754/reconnect_failure.txt)

andreaslundeen, 2024-03-07:
> Hi @andreaslundeen , Kindly use the attahced patch. Please let me know if any addtional information is required. [reconnect_failure.txt](https://github.com/espressif/esp-nimble/files/14522754/reconnect_failure.txt)

error: patch failed: nimble/host/src/ble_gap.c:5808
error: nimble/host/src/ble_gap.c: patch does not apply

Please note that this is based on ESP IDF 4.4.7

darshan7patel, 2024-03-07:
Please kindle check the below:
[reconnect_patch.txt](https://github.com/espressif/esp-nimble/files/14523837/reconnect_patch.txt)

andreaslundeen, 2024-03-07:
> Please kindle check the below: [reconnect_patch.txt](https://github.com/espressif/esp-nimble/files/14523837/reconnect_patch.txt)

Sorry, still errors.

git apply -v --check reconnect_patch.txt
Checking patch nimble/host/src/ble_gap.c...
error: while searching for:
    memcpy(bhc_peer_addr.val, peer_addr->val, BLE_DEV_ADDR_LEN);

#if MYNEWT_VAL(BLE_HOST_BASED_PRIVACY)
    struct ble_hs_resolv_entry *rl = NULL;
    rl = ble_hs_resolv_list_find(bhc_peer_addr.val);

    if (rl != NULL && rl->rl_isrpa) {
        memcpy(bhc_peer_addr.val, rl->rl_peer_rpa, BLE_DEV_ADDR_LEN);
        bhc_peer_addr.type = rl->rl_addr_type;
    }
#endif

error: patch failed: nimble/host/src/ble_gap.c:5372
error: nimble/host/src/ble_gap.c: patch does not apply

rahult-github, 2024-03-07:
Hi @andreaslundeen ,  please try this patch.  I think, the previous patch was generated from tip of the tree which results in the failure at your end.  This should now work. 
[reconnect_patch_v2.txt](https://github.com/espressif/esp-nimble/files/14524923/reconnect_patch_v2.txt)

andreaslundeen, 2024-03-08:
> Hi @andreaslundeen , please try this patch. I think, the previous patch was generated from tip of the tree which results in the failure at your end. This should now work. [reconnect_patch_v2.txt](https://github.com/espressif/esp-nimble/files/14524923/reconnect_patch_v2.txt)

With this latest patch file (including my suggested guard) it works.

andreaslundeen, 2024-03-12:
When will the fix part of an 4.4 bugfix release?

rahult-github, 2024-03-13:
Hi @andreaslundeen , 

 Yes, the change has been merged internally and backported to release branches too. You should see it on github soon once github syncs .

rahult-github, 2024-03-14:
Change is available on github . Closing issue.

closed 2024-03-14 by rahult-github