espressif/esp-nimble #69: Add GATT busy checks

tannewt, 2024-05-30:
I've triggered a bug where I'm sending multiple GATT operations at once. I'm doing an MTU exchange request and a discover services request at once. Other BLE stacks I've used will return a BUSY error for the second request. Nimble doesn't have these checks. Is this by design or would a PR be accepted to add these?

rahult-github, 2024-05-31:
Hi @tannewt , 
 
 >  Is this by design or would a PR be accepted to add these?
 
 Since esp-nimble is taken from upstream mynewt nimble, request to please raise one PR [upstream](https://github.com/apache/mynewt-nimble) , so it can be taken up in all upcoming releases too going forward.

tannewt, 2024-05-31:
Done! https://github.com/apache/mynewt-nimble/issues/1790

closed 2024-05-31 by tannewt