espressif/esp-nimble #70: Changes not vetted upstream?

h2zero, 2024-06-15:
Let me preface this by saying that I have been watching and involved in this fork for years, just look at the history.

Now for the question, why have none of the changes made here since @prasad-alatkar left the scene been PR'd and vetted upstream with the mynewt team? There has been some good changes that would benefit the upstream repo as well, some others not so much. 

Do what you want here, but it would have been nice to see the old ways continue, at least for quality assurance.

rahult-github, 2024-06-15:
Hello @h2zero ,

There have been indeed quite a few fixes which we have pushed upstream and are part of mynewt nimble too .

Some fixes are specific to IDF primarily related to different chipsets , which we cannot move upstream.

But for fixes which are related to Bluetooth as a technology OR any change that is related to C code ( compilation issues , code style etc) , we do push them upstream .

Few of our changes have not been accepted upstream yet with open PRs

Having said that, we would be glad to push any changes that you can point , that we have missed. Pushin changes upstream works even for us as migration becomes easier for us too going ahead and hence we too are aligned to ensure as many changes as we can push upstream .

h2zero, 2024-06-16:
My apologies @rahult-github, yes there have been quite a few changes pushed upstream, thank you for that, they just weren't always easy to spot, especially when tired :).

 I've been bouncing around between the repos and adjusting my library trying to maintain it in line with this one and some of the changes, especially the porting layer, have been difficult to work around. I guess I was mostly concerned with things like the cpfd addition which broke my library and hasn't been accepted upstream yet.

Anyway sorry to bother, my mistake that I didn't see the PR's upstream.

closed 2024-06-16 by h2zero