espressif/esp-nimble #71: Get the address of bonded devices

Steven35700, 2024-07-10:
Hi,

I would like to recover all the addresses of the bonded devices in order to integrate them into the whitelist.
The objective of this is that only whitelisted, and therefore bonded, devices can scan and connect to the advertise.
I haven't found any associated functions to do this, do you potentially have another approach to do this ?

Thanks

rahult-github, 2024-07-11:
[ble_store_util_bonded_peers](https://gitlab.espressif.cn:6688/espressif/esp-nimble/-/blob/nimble-1.6.0-idf/nimble/host/src/ble_store_util.c#L81)  API exists which gives the list of devices which are bonded. You can have a wrapper function, where you first get the list of bonded peers and then perform action on it as per your use case.

Steven35700, 2024-07-15:
It works, thank you !

closed 2024-07-15 by Steven35700