espressif/esp-nimble #73: GATT CONN FAIL ESTABILISH on esp32 C3

adlemx, 2024-08-08:
I run the standard example from esp-idf-v5.3\examples\bluetooth\nimble\bleh when I try to connect to esp from the nRF connect, it immediately after the connection starts disconnects with the Error 62 (0x3e): GATT CONN FAIL ESTABLISH

I use ESP32 C3 super mini and esp-idf 5.3

Logs from nrf connect:
```
nRF Connect, 2024-08-07
BLEHR_SENSOR_1.0 (48:XX:XX:XX:XX:XX)
V 23:57:18.108 Connecting to 48:XX:XX:XX:XX:XX...
D 23:57:18.109 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 23:57:18.575 [Callback] Connection state changed with status: 62 and new state: DISCONNECTED (0)
E 23:57:18.575 Error 62 (0x3e): GATT CONN FAIL ESTABLISH
I 23:57:18.575 Disconnected
```

rahult-github, 2024-08-09:
Hi @adlemx , 

 Error code 3E, implies connection failed due to sync issues.  
 
 CONNECTION FAILED TO BE ESTABLISHED /
SYNCHRONIZATION TIMEOUT (0x3E)
The Connection Failed to be Established / Synchronization Timeout error code
indicates that the LL initiated a connection or initiated synchronization to
periodic advertising but the connection has failed to be established or the Link
Layer failed to synchronize with the periodic advertising within 6 periodic
advertising events of the first attempt.

 So, this can be due to environment reasons .  However, you can try to enable  BT_NIMBLE_ENABLE_CONN_REATTEMPT in menuconfig, which basically will attempt to re-connect if disconnection is due to reason 0x3E.  
 
 Also the value BT_NIMBLE_MAX_CONN_REATTEMPT can be changed to decide how many times this re-attempt for connection is to be done, which can be increased as per user's need.  
 
 So please give this a try.

adlemx, 2024-08-09:
Thanks for the answer. I already had BT_NIMBLE_ENABLE_CONN_REATTEMPT enabled. I tried setting BT_NIMBLE_MAX_CONN_REATTEMPT to 255, but there was no change. I also tried from another phone, and there was a different error (Error 133 (0x85): GATT ERROR), or is it the same thing?

rahult-github, 2024-08-09:
Hello @adlemx ,  

So you tried for 255 times for same result ? May i request you to please share esp device side console logs ?

adlemx, 2024-08-09:
@rahult-github, Yes, of course, here are the logs from the ESP, but it seems to me that there is nothing useful in it because during connection attempts nothing new appears in the logs

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c98c
0x4004c98c: usb_uart_rx_one_char in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e4c
entry 0x403cc71a
I (24) boot: ESP-IDF v5.3 2nd stage bootloader
I (24) boot: compile time Aug  9 2024 12:57:27
I (25) boot: chip revision: v0.4
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (37) boot.esp32c3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=10be4h ( 68580) map
I (103) esp_image: segment 1: paddr=00020c0c vaddr=3fc92a00 size=02890h ( 10384) load
I (106) esp_image: segment 2: paddr=000234a4 vaddr=40380000 size=0cb74h ( 52084) load
I (119) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=4d7ech (317420) map
I (169) esp_image: segment 4: paddr=0007d814 vaddr=4038cb74 size=05ca0h ( 23712) load
I (180) boot: Loaded app from partition at offset 0x10000
I (180) boot: Disabling RNG early entropy source...
V (192) mmap: after coalescing, 1 regions are left
I (192) cpu_start: Unicore app
D (192) cpu_start: Pro cpu up
D (200) clk: RTC_SLOW_CLK calibration value: 3509261
V CACHE_ERR: access error intr clr & ena mask is: 0x3f
V CACHE_ERR: illegal error intr clr & ena mask is: 0x23
D (209) cpu_start: calling init function: 0x42000236 on core: 0
0x42000236: __esp_system_init_fn_init_efuse_check at C:/Espressif/frameworks/esp-idf-v5.3/components/efuse/src/esp_efuse_startup.c:31

D (213) cpu_start: calling init function: 0x42000954 on core: 0
0x42000954: __esp_system_init_fn_init_show_cpu_freq at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_system/startup_funcs.c:58

I (219) cpu_start: Pro cpu start user code
I (224) cpu_start: cpu freq: 160000000 Hz
D (228) cpu_start: calling init function: 0x420000e0 on core: 0
0x420000e0: __esp_system_init_fn_init_show_app_info at C:/Espressif/frameworks/esp-idf-v5.3/components/esp_app_format/esp_app_desc.c:104

I (234) app_init: Application information:
I (239) app_init: Project name:     blehr
I (244) app_init: App version:      v5.3
I (248) app_init: Compile time:     Aug  9 2024 12:57:20
I (254) app_init: ELF file SHA256:  3cf771639...
I (260) app_init: ESP-IDF:          v5.3
D (264) cpu_start: calling init function: 0x42000248 on core: 0
0x42000248: __esp_system_init_fn_init_efuse_show_app_info at C:/Espressif/frameworks/esp-idf-v5.3/components/efuse/src/esp_efuse_startup.c:41

I (270) efuse_init: Min chip rev:     v0.3
I (275) efuse_init: Max chip rev:     v1.99
I (280) efuse_init: Chip rev:         v0.4
D (285) cpu_start: calling init function: 0x420026f8 on core: 0
0x420026f8: __esp_system_init_fn_init_heap at C:/Espressif/frameworks/esp-idf-v5.3/components/heap/heap_caps_init.c:25

V (291) memory_layout: reserved range is 0x3c060b2c - 0x3c060b4c
D (297) memory_layout: Checking 5 reserved memory ranges:
D (302) memory_layout: Reserved memory range 0x3fc80000 - 0x3fc92a00
[... 457 lines cut ...]
D (1765) NimBLE: 0x60
D (1775) NimBLE: 0x00
D (1775) NimBLE: 0x00
D (1775) NimBLE: 0x00
D (1775) NimBLE: 0x00
D (1775) NimBLE: 0x00
D (1785) NimBLE: 0x00
D (1785) NimBLE: 0x00
D (1785) NimBLE: 0x00
D (1785) NimBLE: 0x00
D (1795) NimBLE: 0x00
D (1795) NimBLE: 0x07
D (1795) NimBLE: 0x00
D (1795) NimBLE:

D (1805) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1

D (1805) NimBLE: 0x0a
D (1815) NimBLE: 0x20
D (1815) NimBLE: 0x01
D (1815) NimBLE: 0x01
D (1815) NimBLE:

I (1825) main_task: Returned from app_main()
```

rahult-github, 2024-08-09:
Hello @adlemx , 

My apologies as i misread the initiate statement 

> when I try to connect to esp from the nRF connect 

 Here nrf is initiator, not esp. So on esp side there will be no connection event received. The reattempt connection feature works if connection is initiated from esp side and connection fails with reason 0x3E. 
 
 Here, central device has to initiate connection again, server side doesn't have any control . In logs also, esp side is advertising continuously.

adlemx, 2024-08-09:
@rahult-github I tried sketches from examples\bluetooth\bluedroid\ble_50\ble50_security_server and from examples\bluetooth\bluedroid\ble\gatt_security_server. The example with ble 5.0 works, but just ble gives the same error. Is it possible to make nimble work only with ble 5.0? Enable BLE 5 feature is enabled in the config menu, but apparently it still uses bluetooth 4.2

rahult-github, 2024-08-09:
@adlemx  , they are different stacks and hence different examples. For equivalent example, you can refer to examples/bluetooth/nimble/bleprph ?

adlemx, 2024-08-09:
@rahult-github This example also does not work, could it be a problem with the quartz on the board? And is it possible to switch the ESP to internal clocking?

rahult-github, 2025-07-29:
0x3E is error code when connection sync fails to happen. This is related to environment too. However, newer IDF releases has code under BT_NIMBLE_ENABLE_CONN_REATTEMPT flag for both central / peripheral role, where either a connection is retried or advertising retried. Please use latest codebase and confirm.