espressif/esp-nimble #76: 1.6.0 breaking Clang-based builds

KaeLL, 2024-09-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

To build correctly

### Actual behavior (suspected bug)

Build breaks

### Error logs or terminal output

```shell
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c:39:34: error: tentative definition of variable with internal linkage has incomplete non-array type 'struct ble_hs_resolv_data' [-Werror,-Wtentative-definition-incomplete-type]
   39 | static struct ble_hs_resolv_data g_ble_hs_resolv_data;
      |                                  ^
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_resolv.c:39:15: note: forward declaration of 'struct ble_hs_resolv_data'
   39 | static struct ble_hs_resolv_data g_ble_hs_resolv_data;
      |               ^

/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:483:39: error: logical not is only applied to the left hand side of this comparison [-Werror,-Wlogical-not-parentheses]
  483 |         if (ble_host_rpa_enabled() && !memcmp(evt.local_rpa, ble_hs_conn_null_addr, 6) == 0) {
      |                                       ^                                                ~~
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:483:39: note: add parentheses after the '!' to evaluate the comparison first
  483 |         if (ble_host_rpa_enabled() && !memcmp(evt.local_rpa, ble_hs_conn_null_addr, 6) == 0) {
      |                                       ^
      |                                        (                                                   )
/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:483:39: note: add parentheses around left hand side expression to silence this warning
  483 |         if (ble_host_rpa_enabled() && !memcmp(evt.local_rpa, ble_hs_conn_null_addr, 6) == 0) {
      |                                       ^
      |                                       (                                               )
```
```

### Steps to reproduce the behavior

Build a Peripheral example, I guess. Not sure, but shouldn't matter.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Fedora 40

### Shell

Bash

rahult-github, 2024-09-20:
Hi @KaeLL , 

Can you please help confirm if attached patch works for you  ?
[suppress_clang_warning.txt](https://github.com/user-attachments/files/17069887/suppress_clang_warning.txt)

KaeLL, 2024-09-20:
It does.

rahult-github, 2025-05-05:
Fixes via 03462b6686cf49bb0bb026461300fec593c6f4c5

closed 2025-05-05 by rahult-github