espressif/esp-nimble #77: Slow data flow after reconnection following a sudden disconnection.

Waganama, 2024-09-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The ESP32 device is able to receive data without disconnection or latency after a sudden disconnection.

### Actual behavior (suspected bug)

After reconnection, the ESP32 device received data with high latency, and after a few transmissions, a disconnection occurred before the transfer was completed.

### Error logs or terminal output

```shell
See additionale context for the log file
```

### Steps to reproduce the behavior

1. Enable and initialize a characteristic for one GATT characteristic on Nimble for an ESP32 device.
2. Start advertising for the device to be discoverable.
3. On the PC, launch a Python script (using Bleak) to connect and send data (500 or more packets of 514 bytes).
4. Disconnect Bluetooth on the PC (disable the Bluetooth button or disconnect the Bluetooth antenna). The transfer and connection are interrupted.
5. Relaunch the same Python script. Here, the data reception on the device side is very slow, and the connection is interrupted after receiving a few data packets.

### Project release version

esp-idf V5.0 release

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10-11

### Shell

other (details in Additional context)

### Additional context

I tried to upgrade my ESP framework to version 5.3, but it was unsuccessful. I also attempted to reconnect to the device using another PC and an Android device, but the ESP32 still exhibited some strange behavior.

Here is the log from the device : 
[log.poc_ble.20240924110311.txt](https://github.com/user-attachments/files/17111568/log.poc_ble.20240924110311.txt)

Here is the sample project i use on ESP32. You will also find the script python file i use in the sample project
[ble_sample.zip](https://github.com/user-attachments/files/17111589/ble_sample.zip)

rahult-github, 2024-09-27:
You can try to increase the connection supervision timeout to see if it helps ?  Also, log just shows connection update happening, but not the parameters used. You can either add more debug logs at your layer OR enable nimble debug logs to check what new parameters are being negotiated. An aggressive parameter can help ensuring packets are always transferred periodically resulting in higher throughput also

Waganama, 2024-09-30:
Hello,

I haven't tried your test yet, but I did try disabling the Host Flow Control option in Nimble through sdkconfig, and since then, the issue hasn't occurred anymore.

Interestingly, I noticed that in the default sdkconfig settings for ESP32-S3 and ESP32-C3, the Host Flow Control option is disabled by default, whereas it is enabled on the ESP32. I’m not entirely sure about the potential impacts of disabling this option, though.

strange969, 2025-09-03:
Closing this issue. If the problem still persists, please feel free to reopen it.