espressif/esp-nimble #78: v1.5.0 does not compile without CONFIG_BT_NIMBLE_SECURITY_ENABLE

bryghtlabs-richard, 2024-10-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm trying to migrate a project from ESP-IDF v5.1.2 to ESP-IDF v5.1.4 (ESP-Nimble 299279a46795a6). I also saw that c0dd77a8355dd82 is needed as well, so I moved to nimble-1.5.0-idf branch.

### Actual behavior (suspected bug)

But it doesn't compile without CONFIG_BT_NIMBLE_SM_SC_LVL, which isn't defined without CONFIG_BT_NIMBLE_SECURITY_ENABLE

### Error logs or terminal output

```shell
In file included from C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/porting/nimble/include/syscfg/syscfg.h:9,
                 from C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/porting/nimble/include/os/os.h:33,
                 from C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:23:
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c: In function 'ble_att_svr_rx_notify':
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/port/include/esp_nimble_cfg.h:851:34: error: 'CONFIG_BT_NIMBLE_SM_SC_LVL' undeclared (first use in this function); did you mean 'CONFIG_BT_NIMBLE_LOG_LEVEL'?
  851 | #define MYNEWT_VAL_BLE_SM_SC_LVL CONFIG_BT_NIMBLE_SM_SC_LVL
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: in expansion of macro 'MYNEWT_VAL_BLE_SM_SC_LVL'
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2936:9: note: in expansion of macro 'MYNEWT_VAL'
 2936 |     if (MYNEWT_VAL(BLE_SM_SC_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/port/include/esp_nimble_cfg.h:851:34: note: each undeclared identifier is reported only once for each function it appears in
  851 | #define MYNEWT_VAL_BLE_SM_SC_LVL CONFIG_BT_NIMBLE_SM_SC_LVL
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: in expansion of macro 'MYNEWT_VAL_BLE_SM_SC_LVL'
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:2936:9: note: in expansion of macro 'MYNEWT_VAL'
 2936 |     if (MYNEWT_VAL(BLE_SM_SC_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c: In function 'ble_att_svr_rx_indicate':
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/port/include/esp_nimble_cfg.h:851:34: error: 'CONFIG_BT_NIMBLE_SM_SC_LVL' undeclared (first use in this function); did you mean 'CONFIG_BT_NIMBLE_LOG_LEVEL'?
  851 | #define MYNEWT_VAL_BLE_SM_SC_LVL CONFIG_BT_NIMBLE_SM_SC_LVL
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/port/include/esp_nimble_cfg.h:18:49: note: in expansion of macro 'MYNEWT_VAL_BLE_SM_SC_LVL'
   18 | #define MYNEWT_VAL(_name)                       MYNEWT_VAL_ ## _name
      |                                                 ^~~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v5.1.4/components/bt/host/nimble/nimble/nimble/host/src/ble_att_svr.c:3027:9: note: in expansion of macro 'MYNEWT_VAL'
 3027 |     if (MYNEWT_VAL(BLE_SM_SC_LVL) >= 2 && !sec_state.encrypted) {
      |         ^~~~~~~~~~
[0/1] Re-running CMake...CMake Warning at C:/Espressif/frameworks/esp-idf-v5.1.4/tools/cmake/git_submodules.cmake:52 (message):
```

### Steps to reproduce the behavior

1. C:\Espressif\frameworks\esp-idf-v5.1.4\examples\bluetooth\nimble\ble_spp\spp_client>idf.py set-target esp32-s3
2. C:\Espressif\frameworks\esp-idf-v5.1.4\examples\bluetooth\nimble\ble_spp\spp_client>idf.py menuconfig # Disable CONFIG_BT_NIMBLE_SECURITY_ENABLE
3. C:\Espressif\frameworks\esp-idf-v5.1.4\examples\bluetooth\nimble\ble_spp\spp_client>idf.py build

### Project release version

v1.5.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

rahult-github, 2024-10-24:
Hi @bryghtlabs-richard ,

May i request you to pull latest release/v5.1 from github and confirm . This issue is not seen on latest codebase anymore.

Attached is my sdkconfig, which i used to compile for esp32s3 on release/v5.1  

If you still observe the issue, request you to :

 1. Ensure no code mismatch is present from your code vs code on github ( i.e. clean pull is done to ensure all changes are pulled).
 2. Please share your sdkconfig , which we can try at our end too and see if there are any difference in configuration.
 
 
[spp_build.txt](https://github.com/user-attachments/files/17501043/spp_build.txt)
[sdkconfig.txt](https://github.com/user-attachments/files/17501048/sdkconfig.txt)

bryghtlabs-richard, 2024-10-24:
Thanks @rahult-github  - I can confirm switching to ESP-IDF release/v5.1 and updating submodules fixes this. We look forward to the next ESP-IDF v5.1.x release.

closed 2024-10-24 by bryghtlabs-richard