espressif/esp-nimble #80: limited discovery is not working anymore because of observer mode

floitsch, 2024-11-27:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When scanning with `.limited` set to true I don't expect to see advertisements that are general.

### Actual behavior (suspected bug)

Advertisements of general discoverability are accepted.

### Steps to reproduce the behavior

In https://github.com/espressif/esp-nimble/commit/9faf23c01538e9acc126807641c5734a83d9b9d4 support for the observer role was added.
In that commit the following lines were added:
https://github.com/espressif/esp-nimble/blob/ca6c6173fca923cfa1966891fc6623fd9f9cb323/nimble/host/src/ble_gap.c#L1783-L1788

Since BLE_ROLE_OBSERVER is enabled by default, this basically breaks the `limited` flag (for which the test is just a few lines later).

https://github.com/espressif/esp-nimble/blob/ca6c6173fca923cfa1966891fc6623fd9f9cb323/nimble/host/src/ble_gap.c#L1795-L1802

Maybe this is the intended behavior, but it is not clear from the description of the observer role flag that it should have an impact on the normal observation of the central. (At least it broke our unit tests)

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Archlinux

### Shell

ZSH

rahult-github, 2024-11-29:
Hi @floitsch  , thanks for pointing this out. We will fix this.

strange969, 2026-09-28:
Sorry for the delayed response. The changes have been merged in [commit c37cb8b0](https://github.com/apache/mynewt-nimble/commit/c37cb8b09b1b6faa50bf416956a032fe966ea454). Feel free to reopen the issue in case of any further query.

closed 2026-09-28 by rahult-github