espressif/esp-nimble #81: Compilation fails with extended advertising enabled.

h2zero, 2024-12-12:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

compile without errors

### Actual behavior (suspected bug)

Will not compile when extended advertising is enabled.

### Error logs or terminal output

```shell
/opt/esp/idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:6922:9: error: implicit declaration of function 'ble_gap_master_set_timer'; did you mean 'ble_gap_master_timer'? [-Werror=implicit-function-declaration]
 6922 |         ble_gap_master_set_timer(duration_ticks);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
      |         ble_gap_master_timer
cc1: some warnings being treated as errors
```

### Steps to reproduce the behavior

Enable extended advertising and not connection reattempt.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

latest

### Shell

Bash

### Additional context

https://github.com/espressif/esp-nimble/commit/275e6820cda2862e636c35b1acecfe67978684c7#r150288200

rahult-github, 2024-12-13:
Hi @h2zero  , thanks for pointing this. Attached patch should help fix it .  

[0001-fix-nimble-Fix-compilation-issue-for-combination-of-.patch.txt](https://github.com/user-attachments/files/18121616/0001-fix-nimble-Fix-compilation-issue-for-combination-of-.patch.txt)

AxelLin, 2024-12-18:
Hi @rahult-github 

This is probably off-topic.
I have a question regarding commit  [61aae403134790e8f325557786d5200d2dce2920](https://github.com/espressif/esp-idf/commit/61aae403134790e8f325557786d5200d2dce2920) ,
you replace BLE_GAP_EVENT_CONNECT by BLE_GAP_EVENT_LINK_ESTAB.
However, does it make sense to access event->connect.status in BLE_GAP_EVENT_LINK_ESTAB?
I suppose it should access event->link_estab.status for BLE_GAP_EVENT_LINK_ESTAB.

rahult-github, 2024-12-18:
Hi @AxelLin ,

Thanks for pointing out.  There is a change done internally to fix this, should get merged soon.

AxelLin, 2024-12-19:
@rahult-github 

Well, I just found the BLE_GAP_EVENT_LINK_ESTAB is not available in upstream mynewt-nimble.
So this is a new event in esp-nimble.
Please document that why BLE_GAP_EVENT_LINK_ESTAB is needed (does it fix any issue?).
Also, I'm wondering when to check BLE_GAP_EVENT_LINK_ESTAB event, when to check BLE_GAP_EVENT_CONNECT event?
i.e. What's the difference of using BLE_GAP_EVENT_LINK_ESTAB v.s. BLE_GAP_EVENT_CONNECT?

rahult-github, 2024-12-20:
Hi @AxelLin ,
 
  Actually none of the existing events too have any documentation, but will add up for this event citing the reason for this event and when to use it.

strange969, 2025-09-03:
Closing this issue as the fix has been merged.