espressif/esp-nimble #94: ble_hci_trans_set_acl_free_cb implicitly declared for ESP32-C5 with ESP-IDF v5.4.1

jaylikesbunda, 2025-05-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The project should compile successfully, as the necessary HCI transport functions for NimBLE should be available and correctly linked for the ESP32-C5 target.

### Actual behavior (suspected bug)

The project fails to compile with the following error in components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:
[image]

### Error logs or terminal output

```shell
x509_crt_bundle.S.obj
[839/1395] Building C object esp-idf/esp_pm/CMakeFiles/__idf_esp_pm.dir/pm_locks.c.obj
[840/1395] Building C object esp-idf/esp_pm/CMakeFiles/__idf_esp_pm.dir/pm_trace.c.obj
[841/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/gpio_glitch_filter_ops.c.obj
[842/1395] Building C object esp-idf/mbedtls/CMakeFiles/__idf_mbedtls.dir/esp_crt_bundle/esp_crt_bundle.c.obj
[843/1395] Linking C static library esp-idf/mbedtls/libmbedtls.a
[844/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/gpio_pin_glitch_filter.c.obj
[845/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/rtc_io.c.obj
[846/1395] Building C object esp-idf/esp_pm/CMakeFiles/__idf_esp_pm.dir/pm_impl.c.obj
[847/1395] Linking C static library esp-idf/esp_pm/libesp_pm.a
[848/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/dedic_gpio.c.obj
[849/1395] Building C object esp-idf/riscv/CMakeFiles/__idf_riscv.dir/instruction_decode.c.obj
[850/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/gpio_flex_glitch_filter.c.obj
[851/1395] Building ASM object esp-idf/riscv/CMakeFiles/__idf_riscv.dir/vectors.S.obj
[852/1395] Building ASM object esp-idf/riscv/CMakeFiles/__idf_riscv.dir/vectors_clic.S.obj
[853/1395] Building C object esp-idf/riscv/CMakeFiles/__idf_riscv.dir/interrupt.c.obj
[854/1395] Building C object esp-idf/riscv/CMakeFiles/__idf_riscv.dir/interrupt_clic.c.obj
[855/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_common.c.obj
[856/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_io.c.obj
[857/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/gpio_etm.c.obj
[858/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_nt35510.c.obj
[859/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_ssd1306.c.obj
[860/1395] Building C object esp-idf/esp_driver_gpio/CMakeFiles/__idf_esp_driver_gpio.dir/src/gpio.c.obj
[861/1395] Linking C static library esp-idf/esp_driver_gpio/libesp_driver_gpio.a
[862/1395] Linking C static library esp-idf/riscv/libriscv.a
[863/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_ops.c.obj
[864/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/src/esp_lcd_panel_st7789.c.obj
[865/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/i2c/esp_lcd_panel_io_i2c_v1.c.obj
[866/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/controller/esp32c5/ble.c.obj
[867/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/i2c/esp_lcd_panel_io_i2c_v2.c.obj
[868/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/btc/core/btc_alarm.c.obj
[869/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/api/esp_blufi_api.c.obj
[870/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/hci_log/bt_hci_log.c.obj
[871/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/btc/core/btc_manage.c.obj
[872/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/controller/esp32c5/bt.c.obj
[873/1395] Building C object esp-idf/esp_lcd/CMakeFiles/__idf_esp_lcd.dir/spi/esp_lcd_panel_io_spi.c.obj
[874/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/btc/profile/esp/blufi/blufi_protocol.c.obj
[875/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/btc/core/btc_task.c.obj
[876/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/allocator.c.obj
[877/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/btc/profile/esp/blufi/blufi_prf.c.obj
[878/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/alarm.c.obj
[879/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/buffer.c.obj
[880/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/fixed_pkt_queue.c.obj
[881/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/fixed_queue.c.obj
[882/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/pkt_queue.c.obj
[883/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/future.c.obj
[884/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/hash_functions.c.obj
[885/1395] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/common/osi/list.c.obj
[... 120 lines cut ...]
1. Initialize a new ESP-IDF project for the esp32c5 target using ESP-IDF v5.4.1.
2. Ensure the Bluetooth stack is enabled with NimBLE as the host and the controller enabled (e.g., via idf.py\ menuconfig or by having the relevant options in sdkconfig.defaults).
Component config -> Bluetooth -> Bluetooth host -> NimBLE
Component config -> Bluetooth -> Bluetooth controller -> Bluetooth controller enabled
3.Attempt to build the project (e.g., idf.py build).

### Project release version

5.4.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

rahult-github, 2025-05-14:
Hi @jaylikesbunda  , So looks like you have enabled controller to host flow control which we do not enable for c5 and hence this got through.    

but nevertheless this issue will be fixed.

rahult-github, 2025-05-14:
This patch should help fix your issue.  The patch needs to be applied in $IDF_PATH/components/bt/host/nimble/nimble path. 

[fix_compilation_issue.txt](https://github.com/user-attachments/files/20206234/fix_compilation_issue.txt)

jaylikesbunda, 2025-05-14:
> This patch should help fix your issue. The patch needs to be applied in $IDF_PATH/components/bt/host/nimble/nimble path.
> 
> [fix_compilation_issue.txt](https://github.com/user-attachments/files/20206234/fix_compilation_issue.txt)

turning host control flow off got it to build thank you! unfortunately the patch got me a linking error with it turned back on though:

```
[1564/1566] Linking CXX executable Ghost_ESP_IDF.elf
FAILED: Ghost_ESP_IDF.elf
C:\WINDOWS\system32\cmd.exe /C "cd . && S:\Espressif\tools\riscv32-esp-elf\esp-14.2.0_20241119\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,-z,muldefs -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C5=0 -Wl,--Map=S:/GhostESP2/Ghost_ESP/build/Ghost_ESP_IDF.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c5.peripherals.ld -T esp32c5.rom.ld -T esp32c5.rom.api.ld -T esp32c5.rom.rvfp.ld -T esp32c5.rom.wdt.ld -T esp32c5.rom.systimer.ld -T esp32c5.rom.version.ld -T esp32c5.rom.phy.ld -T esp32c5.rom.coexist.ld -T esp32c5.rom.net80211.ld -T esp32c5.rom.pp.ld -T esp32c5.rom.newlib.ld -T esp32c5.rom.newlib-normal.ld -T esp32c5.rom.heap.ld -T memory.ld -T sections.ld @CMakeFiles\Ghost_ESP_IDF.elf.rsp -o Ghost_ESP_IDF.elf && cd ."
S:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/bt/libbt.a(ble_hs_flow.c.obj): in function `ble_hs_flow_startup':
S:/Espressif/frameworks/esp-idf-v5.4.1/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:269:(.text.ble_hs_flow_startup+0xa2): undefined reference to `ble_transport_register_put_acl_from_ll_cb'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

rahult-github, 2025-05-14:
Hi @jaylikesbunda , my apologies. I checked with team internally and controller to host flow control is not present for newer chips. Hence such issues are seen. 

 I will fix the Kconfig.in to address this.

rahult-github, 2025-07-30:
Changes [merged](https://github.com/espressif/esp-idf/commit/ebd4caf3609652245cd0f526f88161de660b5275)  and related backports created.

closed 2025-07-30 by rahult-github