espressif/esp-nimble #103: ble_l2cap_coc_cleanup_chan corrupts os_mbuf pool

psilva-dsr, 2025-06-09:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

when L2CAP COC connection closes, only mbuf owned by the stack should be freed

### Actual behavior (suspected bug)

when L2CAP COC connection closes, `ble_l2cap_coc_cleanup_chan()` frees all mbufs in `chan->coc_rx.sdus`
some of them may have been returned to application layer

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

enable `MYNEWT_VAL_OS_MEMPOOL_POISON` in build
create L2CAP COC connection
exchange some data
close connection

### Project release version

esp-nimble port 1.6.0

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

ESP32

### Shell

other (details in Additional context)

### Additional context

`coc_rx.sdus[]` is managed by `coc_rx.current_sdu_idx` and `coc_rx.next_sdu_alloc_idx`
but `ble_l2cap_coc_cleanup_chan()` does not check them

a simple fix that I am using is, in `ble_l2cap_coc_rx_fn()`, replace
`rx_sdu = NULL;`
with 
`rx->sdus[chan->coc_rx.current_sdu_idx] = NULL;`
this removes the reference to the mbuf from the stack just before returning it to the application

rahult-github, 2025-06-09:
Hi @psilva-dsr , this code is taken as is from upstream nimble code.  I suggest to raise issue [here](https://github.com/apache/mynewt-nimble)

psilva-dsr, 2025-06-09:
done
https://github.com/apache/mynewt-nimble/issues/2045