espressif/esp-nimble #105: Compilation error with GATT caching enabled in server-only configuration (ESP-IDF v5.4.2)

niklas-sr1, 2025-07-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The project should compile successfully with GATT caching enabled in server-only configurations, as this is a valid use case.

### Actual behavior (suspected bug)

ESP-IDF v5.4.2 fails to compile when `CONFIG_BT_NIMBLE_GATT_CACHING=y` is enabled in a server-only BLE configuration (where `CONFIG_BT_NIMBLE_GATT_CLIENT` is not enabled).

### Error logs or terminal output

```shell
/home/niklas/esp/v5.4.2/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c: In function 'ble_gatts_notify_custom':
/home/niklas/esp/v5.4.2/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:5074:10: error: implicit declaration of function 'ble_gatts_check_conn_aware'; did you mean 'ble_gattc_cache_conn_update'? [-Wimplicit-function-declaration]
 5074 |     rc = ble_gatts_check_conn_aware(conn_handle, &aware);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |          ble_gattc_cache_conn_update
```

### Steps to reproduce the behavior

1. Configure ESP-IDF v5.4.2 project with BLE server-only configuration
2. Enable `CONFIG_BT_NIMBLE_GATT_CACHING=y`
3. Ensure `CONFIG_BT_NIMBLE_GATT_CLIENT` is not set
4. Run idf.py build
5. Compilation fails with the error above

### Project release version

ESP-IDF 5.4.2 (NimBLE commit 0225775e49669cd4f070400908f73718bddce83b)

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Manjaro (Arch Linux)

### Shell

Bash

### Additional context

**Root Cause Analysis**: The function ble_gatts_check_conn_aware() is defined with conditional compilation guard `#if MYNEWT_VAL(BLE_GATT_CACHING)` but is being called from code that may have different conditional compilation context after the refactoring in commit `45352c459af99759de65076a8327a1c865bb2a4e`.

**Configuration that triggers the issue** (`sdkconfig`):
```
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_GATT_SERVER=y
CONFIG_BT_NIMBLE_GATT_CACHING=y
# CONFIG_BT_NIMBLE_GATT_CLIENT is not set (server-only configuration)
```

**Workaround**: Temporarily enable `CONFIG_BT_NIMBLE_GATT_CLIENT=y` even if client functionality is not needed (e.g. enable BLE Central role in ESP-IDF config).

**Related Commits**: Issue likely introduced by: 45352c459af99759de65076a8327a1c865bb2a4e ("fix(nimble): Add support for minimilalistic build sdkconfig")

This issue affects any ESP-IDF v5.4.2 project using a BLE server-only configuration with GATT caching enabled. The error suggests that the function definition and usage have mismatched conditional compilation guards after the build system refactoring.

_Co-authored by GitHub Copilot (Claude Sonnet 4)_

niklas-sr1, 2025-07-04:
Hello @rahult-github,
thank you for your response and really quick turnaround time!

I have applied the patch locally and can confirm that the build now runs successfully.

Now that I can run the code on device I have noticed some other runtime issues (I am currently updating our project from ESP-IDF 5.3.1 to 5.4.2) that prevents a transition to 5.4.2.
Though I have to investigate whether these are issues on our side. I will open a further issue if necessary.

Thanks for your help so far!

strange969, 2025-09-04:
Closing this issue as the fix has been merged.

closed 2025-09-04 by rahult-github