espressif/esp-nimble #106: BLE_HS_ENOMEM After Number of Indications Exceeds BT_NIMBLE_GATT_MAX_PROCS

ryancog, 2025-07-08:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm running on ESP-IDF v5.4.2 (the latest version I see), and am experiencing an issue that I'm not entirely certain how to track down.

I have an esp32s3 configured as a BLE peripheral, connected to a phone application functioning as the central. I utilize a write/indicate model for accomplishing asynchronous esp32s3-side processing and data fetching, and so heavily utilize indications.

For one part of my project, I write a new value (a new name for the device), the device (esp32s3) is meant to update this value (which involves ble_svc_gap_device_name_set and an NVS store), and then ble_gatts_indicate after it has been updated.

The application subscribes to the indications as soon as it connects, and so upon receiving this new indication, it updates the value displayed in the UI.

### Actual behavior (suspected bug)

Recently, upon a characteristic being written and the esp32s3 code completing the action and proceeding to indicate, the indication never reaches the phone application, and if I try to write trigger the action after this (accomplished via manual input in the phone app), after I try BT_NIMBLE_GATT_MAX_PROCS times, ble_gatts_indicate_custom() returns BLE_HS_ENOMEM. With NimBLE debug logging enabled, this final failed indication does not result in a "GATT procedure initiated" log.

Looking through the ble source and based on the behavior, my best guess is that it fails at `ble_gattc_proc_alloc` because, for whatever reason, those indications aren't being processed by NimBLE.

I'm not sure how best to troubleshoot this however, as I'm not intimately familiar with the NimBLE source.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

My application is not trivial at this point, and I'm currently working on trying to set up a somewhat workable MRE. For completion's sake here's the general outline:

1. Connect to esp32s3 running NimBLE with phone application (via flutter_blue_plus)
2. Subscribe to indications for name characteristic
3. Trigger name characteristic write with new value via application UI
- Indication is not received
4. Repeat until ENOMEM is returned.

I can share more information as requested, although I'm not sure what may be precisely useful.

### Project release version

ESP-IDF v5.4.2

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

OpenSUSE Slowroll

### Shell

ZSH

rahult-github, 2025-07-08:
Hi @ryancog  , 

Can you please try increasing the value of  NIMBLE_GATT_MAX_PROCS from [here](https://github.com/espressif/esp-idf/blob/master/components/bt/host/nimble/Kconfig.in#L406) during compilaiton and see if it helps your case.

ryancog, 2025-07-08:
Quick update: it was a quirk of my app it appeared the indications were not received. 

Both in the app and doing some fiddling with nRF Connect I can see that the initial indication is received, as are subsequent indications up until ENOMEM is returned, and no further indications are sent.

ryancog, 2025-07-08:
I have assembled a minimum reproducible example. After connecting with nRF Connect app and subscribing to indications, the full serial output is as follows:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x51c
load:0x403c8700,len:0xf08
load:0x403cb700,len:0x31d4
entry 0x403c89fc
I (73) esp_psram: Found 2MB PSRAM device
I (73) esp_psram: Speed: 80MHz
I (74) cpu_start: Unicore app
I (74) cpu_start: Single core mode
I (90) cpu_start: Pro cpu start user code
I (90) cpu_start: cpu freq: 240000000 Hz
I (90) app_init: Application information:
I (93) app_init: Project name:     main
I (97) app_init: App version:      interact-core-34-g2b12d12-dirty
I (104) app_init: Compile time:     Jul  8 2025 13:55:23
I (110) app_init: ELF file SHA256:  87f1f59db...
I (115) app_init: ESP-IDF:          v5.4.2
I (120) efuse_init: Min chip rev:     v0.0
I (125) efuse_init: Max chip rev:     v0.99 
I (130) efuse_init: Chip rev:         v0.2
I (135) heap_init: Initializing. RAM available for dynamic allocation:
I (142) heap_init: At 3FCA3788 len 00045F88 (279 KiB): RAM
I (148) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (154) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (160) heap_init: At 600FE04C len 00001F8C (7 KiB): RTCRAM
I (166) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (174) spi_flash: detected chip: generic
I (178) spi_flash: flash io: qio
I (183) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (190) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (197) nvs_sec_provider: NVS Encryption - Registering HMAC-based scheme...
I (216) pm: Frequency switching config: CPU_MAX: 240, APB_MAX: 80, APB_MIN: 40, Light sleep: DISABLED
I (216) coexist: coex firmware version: 7b9a184
I (220) coexist: coexist rom version e7ae62f
I (225) main_task: Started on CPU0
I (225) main_task: Calling app_main()
I (235) nvs: NVS partition "nvs" is encrypted.
I (235) BLE_INIT: BT controller compile version [6cfabcd]
I (235) BLE_INIT: Bluetooth will use main XTAL as Bluetooth sleep clock.
I (245) BLE_INIT: Feature Config, ADV:1, BLE_50:1, DTM:1, SCAN:1, CCA:0, SMP:1, CONNECT:1
I (255) BLE_INIT: Bluetooth MAC: 98:3d:ae:f0:b1:ba
I (265) phy_init: phy_version 701,f4f1da3a,Mar  3 2025,15:50:10
I (285) BLE Test: Host Task Started.
I (305) NimBLE: GAP procedure initiated: stop advertising.

I (305) BLE Test: Synced
I (305) BLE Test: Device Address: BA:B1:F0:AE:3D:98
I (305) NimBLE: GAP procedure initiated: advertise; 
I (315) NimBLE: disc_mode=2
I (315) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=1216 adv_itvl_max=2056
I (325) NimBLE: 

I (325) BLE Test: Advertising started!
I (4285) NimBLE: GAP procedure initiated: 
I (4285) NimBLE: connection parameter update; conn_handle=1 itvl_min=12 itvl_max=12 latency=0 supervision_timeout=72 min_ce_len=0 max_ce_len=0
I (4295) NimBLE: 

BLE assert llc_con_upd.c 295, param 00000004 00000003
I (7785) NimBLE: GATT procedure initiated: indicate; 
I (7785) NimBLE: att_handle=3

I (7785) BLE Test: Indication completed with: 0
I (9285) NimBLE: GATT procedure initiated: indicate; 
I (9285) NimBLE: att_handle=3
[... 321 lines cut ...]

    res = ble_gatts_add_svcs(GATT_SERVICES);
    assert(res == 0);

    // GAP Initialization
    ble_svc_gap_init();

    res = ble_svc_gap_device_name_set("Test Device");
    assert(res == 0);

    // See https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v4.x.x/doc/html/group___b_l_e___a_p_p_e_a_r_a_n_c_e_s.html
    res = ble_svc_gap_device_appearance_set(APPEARANCE_VAL);
    assert(res == 0);

    xTaskCreateStaticPinnedToCore(hostTask, "NimBLE Host", hostStack.size(), nullptr, 5, hostStack.data(), &hostTaskData, 0);

    while (true) {
        if (charShouldIndicate) {
            res = ble_gatts_indicate(connHandle, charHandle);
            ESP_LOGI(TAG, "Indication completed with: %d", res);
        }
        vTaskDelay(150);
    }
}
```

ryancog, 2025-07-09:
@rahult-github please let me know if you are able to reproduce this issue. I can additionally provide my sdkconfig settings if they seem to be relevant.

ryancog, 2025-07-09:
@rahult-github That log consistently shows upon first connection to my iOS device. I had not noticed it before, but it appears in my real esp32s3 code output as well.

However, it does not appear when connecting to the Android device, and the same indication issue presents itself regardless of which device is connected, so it seems unrelated.

ryancog, 2025-07-09:
Here I have my sdkconfig I used for the MRE

[sdkconfig.txt](https://github.com/user-attachments/files/21147028/sdkconfig.txt)

ryancog, 2025-07-15:
Hi @rahult-github,

Could you let me know if you're able to reproduce this issue, or if there's more information you need from me? Are there sections of the code you would recommend I investigate to try and find the source of this issue? TIA

rahult-github, 2025-07-15:
The code changes you provided are giving compilation issues .. I guess they can't be used as is. However , i think i understand what issue you are facing. Will check on it.

ryancog, 2025-07-15:
I appreciate you checking.

For the code, perhaps I overlooked something, that code should be identical (direct copy/past) to what I uploaded. I'll definitely take another look and see if I missed something.

ryancog, 2025-07-24:
Hello @rahult-github, any luck on this? Any leads I can chase down on my end?

ryancog, 2025-08-11:
Good to know. 

I'm planning on revisiting this again, I've just been busy on other elements of this project. I thought I had tested older IDF versions, but perhaps I had not. 

My suspicion is that it has to do with something waiting for the central acknowledgement that doesn't get cleared properly, but perhaps not and Notifications have the same issue.

qianchenzhumeng, 2025-08-26:
I'm also having the same problem with ESP-IDF v5.5.0.

`CONFIG_BT_NIMBLE_GATT_MAX_PROCS` is 4. After 4 times, `ble_gatts_indicate_custom` will return `BLE_HS_ENOMEM `.

log:

```
I (10442) NimBLE: GATT procedure initiated: indicate; 
I (10442) NimBLE: att_handle=18

I (10442) NimBLE_GATT_Server: measure indication sent!
I (12442) NimBLE: GATT procedure initiated: indicate; 
I (12442) NimBLE: att_handle=18

I (12442) NimBLE_GATT_Server: measure indication sent!
I (14442) NimBLE: GATT procedure initiated: indicate;
I (14442) NimBLE: att_handle=18

I (14442) NimBLE_GATT_Server: measure indication sent!
I (16442) NimBLE: GATT procedure initiated: indicate;
I (16442) NimBLE: att_handle=18

I (16442) NimBLE_GATT_Server: measure indication sent!
I (18442) NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=18 status=6 is_indication=1
I (18442) NimBLE_GATT_Server: ble_gatts_indicate_custom return error(code: 6)
I (20442) NimBLE_GATT_Server: notify event; conn_handle=1 attr_handle=18 status=6 is_indication=1
I (20442) NimBLE_GATT_Server: ble_gatts_indicate_custom return error(code: 6)
```

qianchenzhumeng, 2025-08-26:
Hello @buganini  @ryancog, I found that this issue is related to the values of `CONFIG_BT_NIMBLE_ROLE_CENTRAL` and `CONFIG_NIMBLE_ROLE_CENTRAL`, their values must be set to 'y', otherwise this issue will occur.

ryancog, 2025-08-26:
@qianchenzhumeng Interesting. Well it’s good to know that’s a workaround for the time being. 

I still haven’t gotten a chance to come back around to this to test myself, but I appreciate you sharing this!

rahult-github, 2025-08-29:
Hi @ryancog , @qianchenzhumeng  ,

 Please help take attached patch that can be applied in $IDF_PATH/components/bt/host/nimble/nimble. for this issue. 

[0001-fix-nimble-Added-code-to-handle-sending-indicate-whe.txt](https://github.com/user-attachments/files/22040709/0001-fix-nimble-Added-code-to-handle-sending-indicate-whe.txt)

Please note, if you send indicate requests say in a loop, back to back,  stack will allocate buffers and queue the request and  process the indications one after another ( i.e. send indication only after once a response is received from client for previous indication). This allocation of buffers is still controlled by max procs configuration value. So , it is application's responsibility to check use case for sending of indications and how many to queue so as to not exceed max procs value. Stack will send one indication to remote client and then once response is recevied , it will free the allocated buffer to free up the procedure information. 

However, the better way is to send a new indicate request after notify_tx for the previous indicate request has been received. This way, we can ensure buffers are always available.

closed 2025-12-31 by rahult-github